ARMmbed/mbed-os-example-tls #11: Stack buffers leading to stack overflow

c1728p9, 2016-08-29:
The main stack for most platforms is 4K. The tls-client example uses ~6K of stack which causes a stack overflow. TLS itself seems to be using nearly 4k and main.cpp is using ~2k.
Some places in main.cpp contributing to this stack usage.
https://github.com/ARMmbed/mbed-os-example-tls/blob/master/tls-client/main.cpp#L270
https://github.com/ARMmbed/mbed-os-example-tls/blob/master/tls-client/main.cpp#L373
https://github.com/ARMmbed/mbed-os-example-tls/blob/master/tls-client/main.cpp#L437
https://github.com/ARMmbed/mbed-os-example-tls/blob/master/tls-client/main.cpp#L470

This is causing the issue reported here:
https://github.com/ARMmbed/mbed-os/issues/2567

c1728p9, 2016-08-29:
@pjbakker, what kind of stack usage would you expect from tls? Are there config options to reduce this?

c1728p9, 2016-08-29:
I created #12 to reduce some of the stack usage by addressing the above points.

pjbakker, 2016-08-30:
@c1728p9 Anywhere from 40k-256k depending on configuration and running ciphersuites / certificates?
There are a lot of config options available in *config.h*

This [article on reducing mbed TLS footprint](https://tls.mbed.org/kb/how-to/reduce-mbedtls-memory-and-storage-footprint) might be a good start?

c1728p9, 2016-08-30:
Hi @pjbakker, are you referring to the stack or the heap? There are many boards with a heap >40K. In mbed-os the main thread's stack is no bigger than 4K for any cortex-m board.

pjbakker, 2016-08-30:
@c1728p9 Sorry.. Was referring to ROM size. Misread your question.
Stack size depends even more on the called function(tree).
Can you elaborate a bit on the functions stacked?

c1728p9, 2016-08-30:
I'm seeing the heavy stack usage on mbed-os-example-tls. I have links above to certain functions which are declaring large objects on the stack. Also inside tls there is heavy stack usage, although I don't have precise locations. I wasn't sure if you had any upper bound for stack usage of TLS as you do for the heap usage.

pjbakker, 2016-08-31:
No, I don't think so.

yanesca, 2016-09-05:
I have some ideas to reduce stack size, but I don't know if we want to do any of these:

Several functions allocate buffers on the stack of MBEDTLS_MPI_MAX_SIZE bytes. This is 1024 by default and it is necessary to have it to be able to perform 4096bit by default. If we remove RSA from the example, we can get away with setting it to as low as ~128.

Our modular multiplication uses a ~750 byte long buffer, this can be reduced to 24 if we set MBEDTLS_MPI_WINDOW_SIZE to 1, but this will affect performance adversely.

c1728p9, 2016-09-05:
What about dynamically allocating the buffer on function entry, and freeing it on function exit? Alternatively, you could dynamically allocate it and attach it to a TLS context object.

yanesca, 2016-09-06:
Yes, it is definitely possible, but doing it is not so straightforward. Changing the examples is one thing, but this would affect the library itself. mbed TLS has a standalone version too and there may be other considerations. 

Only @pjbakker and @sbutcher-arm can tell if we can do this change, and even if they decide that we want to do it, it may not happen for a long time, because we are so short on resources.

That is why I wanted to come up with some solutions that do not need changes in the library.

sbutcher-arm, 2016-09-13:
Fix has been merged. Issue can now be closed.

closed 2016-09-13 by sbutcher-arm