ARMmbed/mbed-os-example-tls #13: mbed deploy downloads 4 times mbed-os ?

adustm, 2016-09-05:
Hello,
I did `git clone https://github.com/ARMmbed/mbed-os-example-tls`
then `cd mbed-os-example-tls`
then` mbed deploy`
I was surprised to see 4 times
```
[mbed] Adding library "xxxxxxxxx\mbed-os" from "https://github.com/ARMmbed/mbed-os/" at rev #d2202045ac8f
```

Is it possible to modify the setup so that it takes mbed-os repository only once ? It is very slow (at least for me, arround 30 minutes)

Kind regards
Armelle

yanesca, 2016-09-05:
Each example is a standalone application and the `mbed deploy` should be used in the subdirectory of the given example.

This does not help if you want to try all of them though, but unfortunately we don't know any better solution. Unfortunately mbed cli can't handle several applications. Hiding all but one example in a TARGET_IGNORE directory and switching them occasionally is the only solution to the problem that comes to mind, and it would look extremely bad in my opinion.

sbutcher-arm, 2016-09-06:
The solution with the current tools would be to have one example per repository, which I don't think we want to do at this time for such small examples. The tools team are aware of the shortcoming so there's no action required there.

As this is a question which I think we've answered, I'm closing the issue. If you disagree, or want to follow up please feel free to do so.

closed 2016-09-06 by sbutcher-arm

andresag01, 2016-09-06:
I know that this question has already been closed, but if you are interested in only one of the examples, you can do the following command sequence to donwload mbed-os only once:

```
$ git clone https://github.com/ARMmbed/mbed-os-example-tls
$ cd mbed-os-example-tls/<EXAMPLE>
$ mbed config root .
$ mbed deploy # This will only set up one example which is significantly faster than all of them...
....
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #d2202045ac8f
....
```

yanesca, 2016-09-06:
It has been done for all of the examples and the results committed to the repository and the ` mbed config root .` command is not necessary.