ARMmbed/mbed-os-example-tls #14: mbedTLS causes stack overflow on some mbedOS targets

andresag01, 2016-09-06:
The stack requirements for mbed TLS largely depend on the configuration used at compile time. However, there are no stack memory consumption figures available to help users decide on an appropriate stack size for embedded applications. As a result, it is very common that mbed TLS causes stack overflow errors that may or may not the caught by the system depending on the severity of the issue.

andresag01, 2017-02-14:
Currently, some of the example applications use the mbed_app.json file to set the macro `OS_MAINSTKSIZE` and increase the stack size. However, I do not think this has the desired effect in the latest mbed OS version. This should be removed from the examples.

RonEld, 2018-06-10:
I believe that #116 would resolve most of the issues raised from the benchmark application

andresag01, 2018-10-22:
Indeed, the stack issues should now be resolved with #116, #194 and other refactoring PRs for authcrypt and tls-client. So this issue can now be closed.

closed 2018-10-22 by andresag01