ARMmbed/mbed-os-example-tls #15: Assertation failed in the benchmark example

toyowata, 2016-09-27:
Following assertion occurred in the benchamrk example.

```
  SHA-256                  :       1910 Kb/s,         61 cycles/byte
  SHA-512                  :        614 Kb/s,        191 cycles/byte
  AES-CBC-128              :       1355 Kb/s,         86 cycles/byte
  AES-CBC-192              :       1191 Kb/s,         98 cycles/byte
  AES-CBC-256              :       1063 Kb/s,        110 cycles/byte
  AES-GCM-128              :        440 Kb/s,        267 cycles/byte
  AES-GCM-192              :        421 Kb/s,        279 cycles/byte
  AES-GCM-256              :        404 Kb/s,        291 cycles/byte
  AES-CCM-128              :        586 Kb/s,        200 cycles/byte
  AES-CCM-192              :        523 Kb/s,        224 cycles/byte
  AES-CCM-256              :        472 Kb/s,        248 cycles/byte
  CTR_DRBG (NOPR)          :       1110 Kb/s,        105 cycles/byte
  CTR_DRBG (PR)            :        796 Kb/s,        147 cycles/byte
  HMAC_DRBG SHA-256 (NOPR) :        218 Kb/s,        541 cycles/byte
  HMAC_DRBG SHA-256 (PR)   :        192 Kb/s,        616 cycles/byte
  RSA-2048                 :      30 ms/ public
  RSA-2048                 :    1041 ms/private
  RSA-4096                 :     100 ms/ public
  RSA-4096                 :    5742 ms/private
  ECDSA-secp384r1          :  mbed assertation failed: interrupts_disabled, file: .\mbed-os\hal\common\mbed_critical.c, line 54
```

### mbed OS version
```
[mbed] Importing program "mbed-os-example-tls" from "https://github.com/armmbed/mbed-os-example-tls" at latest revision in the current branch
[mbed] Adding library "authcrypt\mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #21dd7008a154
[mbed] Adding library "benchmark\mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #21dd7008a154
[mbed] Adding library "hashing\mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #21dd7008a154
[mbed] Adding library "tls-client\mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #21dd7008a154
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
```

### Build command

```
$ cd benchmark
$ mbed compile -m K64F -t GCC_ARM
```

mrudu172, 2016-09-30:
I am also facing the exact same issue. Please post here if any solution is found!
Thank you.

RonEld, 2018-08-01:
This is an old issue, and I believe it was resolved with https://github.com/ARMmbed/mbed-os-example-tls/commit/83dd1377242a515821765e6c01ea9abac7cdac75#diff-11d19f91de60e62a1952ab3e8b07cd0c

sbutcher-arm, 2018-10-18:
If it wasn't already fixed, it's now fixed by PR #194 which has now been merged. Issue can be closed.

closed 2018-10-18 by sbutcher-arm