ARMmbed/mbed-os-example-tls #25: mbed TLS examples fail in NUMAKER_PFM_NUC472 board

andresag01, 2016-11-30:
Even though the examples compile correctly with ARMCC and gcc, none of them shows the expected behaviour when the applications are flahed onto the target and the serial terminal is connected.

To reproduce this, compile any of the examples with mbed-os-5.3.0-rc1:
```
mbed compile -t GCC_ARM -m NUMAKER_PFM_NUC472
```
Then flash the device and open the serial terminal.

0xc0170, 2016-12-01:
cc @ccli8

yanesca, 2016-12-01:
The same happens when compiled with the online IDE.

ccli8, 2016-12-02:
Hi, I cannot re-produce the issues.
### My environment
**mbed-os**
mbed-os-5.3.0-rc1
**mbed-os-example-tls**
ff1464123b9619c8e746b91a2458b6a8fd037705

### My test result
**mbed-os-example-tls/authcrypt (NUMAKER_PFM_NUC472, ARM)**
```
plaintext message: 536f6d65207468696e67732061726520626574746572206c65667420756e7265616400
ciphertext: edd9ff61518a38b0f6a878d703c78d3653003e7c4d21e4cb7030daf89e24e9237ecb691b831af357e8c8fc7dc62791f18fe69ae39994bdf65724f3
decrypted: 536f6d65207468696e67732061726520626574746572206c65667420756e7265616400

DONE
```

### mbed-os-example-tls/benchmark (NUMAKER_PFM_NUC472, ARM)
```
  SHA-256                  :        959 Kb/s,         88 cycles/byte
  SHA-512                  :        465 Kb/s,        183 cycles/byte
  AES-CBC-128              :        129 Kb/s,        664 cycles/byte
  AES-CBC-192              :        107 Kb/s,        798 cycles/byte
  AES-CBC-256              :         92 Kb/s,        931 cycles/byte
  AES-GCM-128              :         85 Kb/s,       1006 cycles/byte
  AES-GCM-192              :         73 Kb/s,       1172 cycles/byte
  AES-GCM-256              :         64 Kb/s,       1339 cycles/byte
  AES-CCM-128              :         50 Kb/s,       1731 cycles/byte
  AES-CCM-192              :         42 Kb/s,       2064 cycles/byte
  AES-CCM-256              :         37 Kb/s,       2398 cycles/byte
  CTR_DRBG (NOPR)          :         88 Kb/s,        971 cycles/byte
  CTR_DRBG (PR)            :         67 Kb/s,       1289 cycles/byte
  HMAC_DRBG SHA-256 (NOPR) :         63 Kb/s,       1371 cycles/byte
  HMAC_DRBG SHA-256 (PR)   :         56 Kb/s,       1566 cycles/byte
  RSA-2048                 :     572 ms/ public
  RSA-2048                 :   19158 ms/private
  RSA-4096                 :    2116 ms/ public
  RSA-4096                 :  121278 ms/private
  ECDHE-secp384r1          :    8848 ms/handshake
  ECDHE-secp256r1          :    4554 ms/handshake
  ECDHE-Curve25519         :    3919 ms/handshake
  ECDH-secp384r1           :    4324 ms/handshake
  ECDH-secp256r1           :    2245 ms/handshake
  ECDH-Curve25519          :    2029 ms/handshake

DONE
```

**mbed-os-example-tls/hashing (NUMAKER_PFM_NUC472, ARM)**
```
Method 1: 315f5bdb76d078c43b8ac0064e4a0164612b1fce77c869345bfc94c75894edd3
Method 2: 315f5bdb76d078c43b8ac0064e4a0164612b1fce77c869345bfc94c75894edd3
Method 3: 315f5bdb76d078c43b8ac0064e4a0164612b1fce77c869345bfc94c75894edd3
Method 4: 315f5bdb76d078c43b8ac0064e4a0164612b1fce77c869345bfc94c75894edd3

DONE
```

### mbed-os-example-tls/tls-client (NUMAKER_PFM_NUC472, ARM)
```
100 full
Link Up
Using Ethernet LWIP
Client IP Address is 192.168.43.170
Connecting with developer.mbed.org
Starting the TLS handshake...
TLS connection to developer.mbed.org established
Server certificate:
    cert. version     : 3
    serial number     : 11:21:B8:47:9B:21:6C:B1:C6:AF:BC:5D:0C:19:52:DC:D7:C3
    issuer name       : C=BE, O=GlobalSign nv-sa, CN=GlobalSign Organization Validation CA - SHA256 - G2
    subject name      : C=GB, ST=Cambridgeshire, L=Cambridge, O=ARM Ltd, CN=*.mbed.com
    issued  on        : 2016-03-03 12:26:08
    expires on        : 2017-04-05 10:31:02
    signed using      : RSA with SHA-256
    RSA key size      : 2048 bits
    basic constraints : CA=false
    subject alt name  : *.mbed.com, mbed.org, *.mbed.org, mbed.com
    key usage         : Digital Signature, Key Encipherment
    ext key usage     : TLS Web Server Authentication, TLS Web Client Authentication
Certificate verification passed

HTTPS: Received 439 chars from server
HTTPS: Received 200 OK status ... [OK]
HTTPS: Received 'Hello world!' status ... [OK]
HTTPS: Received message:

HTTP/1.1 200 OK
Server: nginx/1.7.10
Date: Fri, 02 Dec 2016 01:56:48 GMT
Content-Type: text/plain
Content-Length: 14
Connection: keep-alive
Last-Modified: Fri, 27 Jul 2012 13:30:34 GMT
Accept-Ranges: bytes
Cache-Control: max-age=36000
Expires: Fri, 02 Dec 2016 11:56:48 GMT
X-Upstream-L3: 172.17.0.3:80
X-Upstream-L2: developer-sjc-indigo-1-nginx
Strict-Transport-Security: max-age=31536000; includeSubdomains

Hello world!
```

c1728p9, 2016-12-02:
Hi @ccli8, it looks like this occurs in Linux, but not in Windows. If you flash on a linux machine you should be able to reproduce this problem.

andresag01, 2016-12-06:
As described by @c1728p9, the problem is related to some issues to flash the board in systems other than Windows. We have retested all the examples using Windows and they work correctly. Since this is not a problem with mbed TLS or the examples, I am closing this issue.

closed 2016-12-06 by andresag01