ARMmbed/mbed-os-example-tls #26: tls-client example failing in UBLOX_EVK_ODIN_W2 dev board

andresag01, 2016-11-30:
The tls-client compiles correctly with ARMCC and gcc for the UBLOX_EVK_ODIN_W2 target, yet no output is observed in the serial terminal when the application is run.

To reproduce, compile the program, flash and then connect the serial terminal:
```
mbed compile -t GCC_ARM -m UBLOX_EVK_ODIN_W2
```

andreaslarssonublox, 2016-12-01:
I tested this and it seems to work for me if the EMAC macro is removed from the device_has in targets.json for UBLOX_EVK_ODIN_W2. If this is not removed I get the same behavior as you have with no output.
The default MAC is now Wi-Fi introduced in this commit:
https://github.com/ARMmbed/mbed-os/commit/20b7f057214a9edcda554c50ebec99e51f9756ed

bulislaw, 2016-12-01:
Shouldn't TLS work on both Ethernet and WiFi?

bulislaw, 2016-12-01:
Ok, synced with Martin -> the issue is by default Ethernet is not enabled on Odin board, with EMAC macro enabled the WiFi is active. The example doesn't support WiFi, therefore we have a fail.

yanesca, 2016-12-01:
It has run successfully when compiled in the online IDE.

andresag01, 2017-01-24:
@0xc0170: Removing `EMAC` from the `device_has` for the target in the targets.json fixes the problem. However, is there a way to force the target to use Ethernet instead of Wifi without modifying mbed OS's targets.json to remove the `EMAC`?

bulislaw, 2017-01-25:
You can use `device_has_remove` in mbed_app.json like in https://github.com/ARMmbed/mbed-os-example-sockets/blob/master/mbed_app.json

andresag01, 2017-01-25:
@bulislaw: Thanks for the information. I have created a PR with the suggested change. https://github.com/ARMmbed/mbed-os-example-tls/pull/41

andresag01, 2017-01-25:
The fix for this issue has been merge...

closed 2017-01-25 by andresag01