ARMmbed/mbed-os-example-tls #28: Unexpected behaviour of mbed export?

andresag01, 2016-12-01:
I believe there is some unexpected behaviour when exporting the examples using the command:
```
mbed export -i make_iar -m K64F
```
It appears that if the `mbed_settings.py` does dont contain an acceptable path for `IAR_PATH` then the generated makefiles do not contain the correct include path. However, I find surprising that even after setting `IAR_PATH` correctly, invocations of the compiler are not prepended with the correct path to the compiler, resulting in build failures. As a workaround, I had to include the path to the compiler in the environment variabled.

0xc0170, 2016-12-01:
probably an issue for mbed-cli ?

andresag01, 2016-12-01:
@0xc0170: Probably, but I am not sure how the build system is structured, there are a number of python scripts as well in mbed-os.

ashok-rao, 2016-12-09:
Issue also seen with NUCLEO_F429ZI:

IAR path set in mbed_settings.py
[image]

```
C:\Ashok\mbed_CLI\v5_3_OoB\TLS\mbed-os-example-tls\tls-client>mbed export -i make_iar -m NUCLEO_F429ZI
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
c:\python27\lib\site-packages\fuzzywuzzy\fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE

C:\Ashok\mbed_CLI\v5_3_OoB\TLS\mbed-os-example-tls\tls-client>make clean
rmdir /S /Q "BUILD"

C:\Ashok\mbed_CLI\v5_3_OoB\TLS\mbed-os-example-tls\tls-client>make
C:/Ashok/mbed_CLI/v5_3_OoB/TLS/mbed-os-example-tls/tls-client/makefile:587: warning: overriding commands for target `.s.o'
C:/Ashok/mbed_CLI/v5_3_OoB/TLS/mbed-os-example-tls/tls-client/makefile:582: warning: ignoring old commands for target `.s.o'
"Compile: main.cpp"
process_begin: CreateProcess(NULL, iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 -Oh --cpu Cortex-M4F --thumb --dlib_config "C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.5/arm\inc\c\DLib_Config_Full.h" --c++ --no_rtti --no_exceptions --guard_calls --no_static_des
truction --guard_calls --no_static_destruction -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DTARGET_NUCLEO_F429ZI -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DTARGET_STM32F4 -D__CMSIS_RTOS
 -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_PORTOUT=1 -DDEVICE_SPI_ASYNCH=1 -DTOOLCHAIN_IAR -DDEVICE_INTERRUPTIN=1 -DTARGET_STM32F429xx -DDEVICE_I2C=1 -DTARGET_STM32F429 -DRTC_LSI=1 -D__CORTEX_M
4 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_PWMOUT=1 -DTARGET_LIKE_MBED -DTARGET_F429_F439 -DTARGET_FF_ARDUINO -DMBED_BUILD_TIMESTAMP=1481282114.58 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DDEVICE_TRNG=1 -DTARGET_STM32F429ZI -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_ERROR_R
ED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_PORTIN=1 -DARM_MATH_CM4 --preinclude=mbed_config.h -I../ -I../. -I.././mbed-os -I.././mbed-os/cmsis -I.././mbed-os/cmsis/TOOLCHAIN_IAR -I.././mbed-os/docs -I.././mbed-os/drivers -I.././mbed-os/events -I.././mbed-os/events/equeue
 -I.././mbed-os/features -I.././mbed-os/features/FEATURE_LWIP -I.././mbed-os/features/FEATURE_LWIP/lwip-interface -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch -I
.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/TARGET_STM32F4 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/TARGET_STM32F4/TARGET_F429_F439 -I.././mbed-os/features/
FEATURE_LWIP/lwip-interface/lwip-sys -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-sys/arch -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/api -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/core -I../
./mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/core/ipv4 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/core/ipv6 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip -I.././mbed-os/featu
res/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip/priv -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/netif -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/netif/ppp -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/netif/ppp/polarssl
 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif -I.././mbed-os/features/frameworks -I.././mbed-os/features/frameworks/greentea-client -I.././mbed-os/features/frameworks/greentea-client/greentea-client -I.././mbed-os/features/frameworks/greentea-client/source -I.././mbed-os/featur
es/frameworks/unity -I.././mbed-os/features/frameworks/unity/source -I.././mbed-os/features/frameworks/unity/unity -I.././mbed-os/features/frameworks/utest -I.././mbed-os/features/frameworks/utest/source -I.././mbed-os/features/frameworks/utest/utest -I.././mbed-os/features/mbedtls -I.././mbed-os/fe
atures/mbedtls/importer -I.././mbed-os/features/mbedtls/inc -I.././mbed-os/features/mbedtls/inc/mbedtls -I.././mbed-os/features/mbedtls/platform -I.././mbed-os/features/mbedtls/platform/inc -I.././mbed-os/features/mbedtls/platform/src -I.././mbed-os/features/mbedtls/src -I.././mbed-os/features/nanos
tack -I.././mbed-os/features/netsocket -I.././mbed-os/features/storage -I.././mbed-os/hal -I.././mbed-os/hal/storage_abstraction -I.././mbed-os/platform -I.././mbed-os/rtos -I.././mbed-os/rtos/rtx -I.././mbed-os/rtos/rtx/TARGET_CORTEX_M -I.././mbed-os/rtos/rtx/TARGET_CORTEX_M/TARGET_RTO [... cut ...]
F439/TARGET_NUCLEO_F429ZI -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_F429_F439/TARGET_NUCLEO_F429ZI/device -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_F429_F439/device -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_F429_F439/device/TOOLCHAIN_IAR -I.././mbed-os/targe
ts/TARGET_STM/TARGET_STM32F4/device -o main.o ../main.cpp, ...) failed.
make (e=2): The system cannot find the file specified.
make[1]: *** [main.o] Error 2
make: *** [all] Error 2

C:\Ashok\mbed_CLI\v5_3_OoB\TLS\mbed-os-example-tls\tls-client>
```

0xc0170, 2016-12-09:
cc @theotherjimmy Is this correct, that makefile does not contain full paths, thus any mbed_settings is ignored? Is that desired behavior?

theotherjimmy, 2016-12-09:
>Is this correct, that makefile does not contain full paths, thus any mbed_settings is ignored? Is that desired behavior?

The short answer is yes.

0xc0170, 2016-12-13:
@andresag01 Please review the answer from @theotherjimmy . This is intended behavior, if not properly documented we should. There's that fix above that was integrated and will be part of the next patch release.

andresag01, 2017-01-24:
@0xc0170 @theotherjimmy: I tried the latest version of mbed-os (#34c1facf42) and the issue persists. That is, if I set the `ARM_PATH` (or `IAR_PATH`) to a correct location in `mbed_settings.py` and then run:
```
mbed export -i make_armc5 -m K64F
```
the build fails because the Makefiles are setup to find the compiler in the `PATH` variable even though I set an absolute path to the compiler. Interestingly enough, if I set `ARM_PATH` correctly and then run:
```
mbed compile -t ARM -m K64F
```
it actually picks up the correct path and compiles without a problem.

I had a look at the the change at ARMmbed/mbed-os#3414 that @theotherjimmy pointed out, but I think this solves a different issue (still related though). Once more, is this the expected behaviour? If it is I will close this bug, if it is not then I will file an issue in mbed-os as this is not related to mbed TLS.

theotherjimmy, 2017-01-25:
@andresag01 The make exporter environment assumes that the compiler is in you `PATH` environment variable, irrespective of what `mbed compile` does.

andresag01, 2017-01-25:
@theotherjimmy: So you mean that this is the expected behaviour and this is actually not a bug?

theotherjimmy, 2017-01-25:
@andresag01 that is correct

andresag01, 2017-01-25:
@theotherjimmy: Thanks for clarifying, I am closing this bug report as it is the desired behaviour.

closed 2017-01-25 by andresag01