ARMmbed/mbed-os-example-tls #33: [Online IDE] [Benchmark]: Compilation fails for Nuvoton NUC472.

ashok-rao, 2016-12-09:
Steps to reproduce:
1. Import example from the "import via URL" link and point to https://github.com/ARMmbed/mbed-os-example-tls.git
2. Change platform to Nuvoton NUC472. Remove other examples from workspace.
3. Compile.
4. Fails with following error.

[image]

andresag01, 2017-01-24:
Notes:
* This issue cannot be reproduced using command line tools (mbed-cli) or the exporter's makefiles (mbed export).
* The compilation error can still be reproduced with the HEAD of master branch in this repo.
* I have raised a bug report in mbed OS ARMmbed/mbed-os#3637. I will not close this bug yet as it is related to the mbed TLS implementation of hardware entropy in mbed OS.

andresag01, 2017-01-26:
The cause of the problem reported here is that the online IDE cannot handle multiple examples under the same project directory. Instead of importing the mbed TLS examples from their GitHub repo, use the following mirrors:
* https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-benchmark
* https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-tls-client
* https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-hashing
* https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-authcrypt

I will add links to these mirrors to the mbed TLS example's documentation.

andresag01, 2017-01-26:
I have created a PR to change the main README.md: https://github.com/ARMmbed/mbed-os-example-tls/pull/42

andresag01, 2017-01-27:
PR has been merged. Closing this issue.

closed 2017-01-27 by andresag01