ARMmbed/mbed-os-example-tls #34: [Exporter] [TLS-Client] : Platform not supported by IAR.

ashok-rao, 2016-12-09:
1. Update repo.
mbed-os (7b974cac6578)
```
C:\Ashok\mbed_CLI\v5_3_OoB\TLS\mbed-os-example-tls\tls-client>mbed export -i IAR -m NUMAKER_PFM_NUC472
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
c:\python27\lib\site-packages\fuzzywuzzy\fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S] [-E] [--source SOURCE_DIR] [-D MACROS]
                  [--profile PROFILE] [--update-packs]
project.py: error: NUMAKER_PFM_NUC472 not supported by iar
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u C:\Ashok\mbed_CLI\v5_3_OoB\TLS\mbed-os-example-tls\tls-client\mbed-os\tools\project.py -i iar -m NUMAKER_PFM_NUC472 --source ." in "C:\Ashok\mbed_CLI\v5_3_OoB\TLS\mbed-os-example-tls\tls-client"
---
```

0xc0170, 2016-12-12:
Hello, can you please file this against mbed-os  (move this issue there), as the support for exporters is there? Thanks

andresag01, 2017-01-24:
This issue is mostly related to the mbed-cli exporter instead of the mbed TLS example code. So I am closing this one and filed a report https://github.com/ARMmbed/mbed-os/issues/3633 in mbed-os

closed 2017-01-24 by andresag01