ARMmbed/mbed-os-example-tls #52: [OOB_5_4] Benchmark program hangs at  "RSA-2048"

soramame21, 2017-03-01:
I exported  **benchmark as "make_iar"**, and built benchmark.bin with no error. 
the program hangs at "RSA-2048" cannot go through as following. but the benchmark.bin built from mbed-cli command with IAR toolchain works well. 

[image]

# How to build the bad execuable

> (workspace) C:\iotBu\ooT-228\import\mbed-os-example-tls\benchmark>mbed export -i
>  make_iar -m NUCLEO_F746ZG -vv > a00 2>&1     //`marker 1`: makefile is generated as a result
> (workspace) C:\iotBu\ooT-228\import\mbed-os-example-tls\benchmark>make clean
> (workspace) C:\iotBu\ooT-228\import\mbed-os-example-tls\benchmark>make
> 

### makefile
I cannot attach the makefile generated at `maker 1`.
if you need it, I can send it by email.

# How to build the working execuable

> (workspace) C:\iotBu\ooT-228\import\mbed-os-example-tls\benchmark>mbed compile -
> m NUCLEO_F746ZG -t IAR -vv > ren 2>&1
> 
> (workspace) C:\iotBu\ooT-228\import\mbed-os-example-tls\benchmark>

### compilation options 
`[DEBUG] Compile: C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.5\arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 -Oh --cpu Cortex-M7 --thumb --dlib_config DLib_Config_Full.h --fpu=VFPv5_sp --vla -DTARGET_STM32F746xG -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DDEVICE_PORTOUT=1 -DUSBHOST_OTHER -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_LOWPOWERTIMER=1 -DTARGET_STM32F7 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DDEVICE_ANALOGOUT=1 -DTARGET_STM32F746ZG -DDEVICE_CAN=1 -DARM_MATH_CM7 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M7 -DDEVICE_TRNG=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M7 -DDEVICE_SPI_ASYNCH=1 -DTOOLCHAIN_IAR -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_STDIO_MESSAGES=1 -D__CORTEX_M7 -DTARGET_STM32F746 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL=1 -DTARGET_NUCLEO_F746ZG -DTARGET_RTOS_M4_M7 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DMBED_BUILD_TIMESTAMP=1488364481.74 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_RTC=1 -DDEVICE_I2C_ASYNCH=1 -f .\BUILD\NUCLEO_F746ZG\IAR\.includes_395d7dfa13e0bc77e08d33db466febf1.txt --preinclude=.\BUILD\NUCLEO_F746ZG\IAR\mbed_config.h --dependencies .\BUILD\NUCLEO_F746ZG\IAR\mbed-os\targets\TARGET_STM\i2c_api.d -l .\BUILD\NUCLEO_F746ZG\IAR\mbed-os\targets\TARGET_STM\i2c_api.s.txt -o .\BUILD\NUCLEO_F746ZG\IAR\mbed-os\targets\TARGET_STM\i2c_api.o .\mbed-os\targets\TARGET_STM\i2c_api.c`

soramame21, 2017-03-07:
@simonb @MarceloSalazar Could you give comment for this issue?

sbutcher-arm, 2017-03-07:
This is similar to #56 which was a heap problem and this is likely an insufficient stack problem, similar to #14.

@mazimkhan - Could you please look at this?

mazimkhan, 2017-03-07:
@soramame21 I am not able to reproduce this issue locally. IAR workbench version is 7.70.1.11486. IAR compiler version is 7.70.1.11437/W32. Can you please provide the version you are using?

soramame21, 2017-03-07:
@mazimkhan I'm using IAR workbench version below.
[image]

mazimkhan, 2017-03-07:
Please try with IAR 7.7

```
   IAR ANSI C/C++ Compiler V7.70.1.11437/W32 for ARM
   Copyright 1999-2016 IAR Systems AB.
   Network license: 10.2.203.155 (STD)
Command line error: Unexpected command line arguments:
            -version

```

soramame21, 2017-03-07:
@mazimkhan @toyowata I'll try it tomorrow, but not sure how to get the specific compiler.

soramame21, 2017-03-08:
@mazimkhan @toyowata I cannot get the `IAR ANSI C/C++ Compiler V7.70.1.11437/W32 for ARM`. I do not change my IAR compiler. but I found the exported makefile including compilation flag **"-On"**, when I changed it to **"-Oh"**, then the program started working.

mazimkhan, 2017-03-08:
```On``` would mean no optimisation. This looks like an exporter bug because by default optimisations are On. 
However, it should be possible to debug the example. Can you try with mbed-os PR 3902 that you used with other issue. Since that corrects both heap and stack sizes.

soramame21, 2017-03-08:
@mazimkhan I had applied the **mbed-os PR 3902** with enlarged heap size already. it is not relevant to this issue.

soramame21, 2017-03-08:
@mazimkhan I got **HardFault_Handler** on IAR Embeded Workbench.

mazimkhan, 2017-03-08:
@soramame21 do you get HardFault for PR 3902 or without or anyways?

soramame21, 2017-03-08:
@mazimkhan I applied  PR 3902 at this morning, then started my test. I didn't remove it. I believe this HardFault cannot be fixed by removing PR 3902.

soramame21, 2017-03-08:
@mazimkhan the HardFault can be reproduced when I choose `Level None or Level Low` at the optimization setting screen.

but the HardFault can be fixed when I choose `Level Medium or Level High`

[image]

soramame21, 2017-03-08:
@mazimkhan I identified that the HardFault can be fixed when I set `Level Medium without options` as following.

[image]

soramame21, 2017-03-14:
@mazimkhan This issue is not fixed, it is reproduced. when I built binary for target **HEXIWEAR**, the compilation flag "-On" is exported to makefile, it should be "-Oh"! 

### build command
```>mbed export -i make_iar -m HEXIWEAR >> bb1 2>&1
>make clean
>make V=1 > ren01 2>&1```

mazimkhan, 2017-03-14:
@theotherjimmy can you please look into the exporter issue in above comment.

soramame21, 2017-03-14:
@mazimkhan could you rephrase your comment? I don't follow you. I had verified following code.
Is it OK?

mbed-os-example-tls\benchmark\mbed-os\targets\TARGET_Freescale\TARGET_MCUXpresso_MCUS\TARGET_MCU_K64F\device\TOOLCHAIN_IAR\MK64FN1M0xxx12.icf
`/* Heap 1/4 of ram and stack 1/8 */
define symbol __stack_size__=0x8000;
define symbol __heap_size__=0x10000;`

mazimkhan, 2017-03-14:
@soramame21 There are two issues here. 
1. The example should work regardless of optimisation level.
2. The exporter should be consistent with other IDEs/build tools that compile with optimisation On.

I have added Jimmy here to look into the exporter issue. Regarding example not working with optimisation low or off, we can put that as a known issue and schedule to fix that.

soramame21, 2017-03-14:
@mazimkhan I see. Thank you for explanation and quick reaction!

theotherjimmy, 2017-03-14:
@soramame21 @mazimkhan see https://github.com/ARMmbed/mbed-os/issues/3890. Exporters use the `debug` profile by default. You can change that by passing `--profile=develop` for the "default" profile into the `mbed export` command.

So:
>The example should work regardless of optimisation level.

Agreed

>The exporter should be consistent with other IDEs/build tools that compile with optimisation On.

They are consistant, with `--profile=debug`.

sbutcher-arm, 2018-10-18:
Fixed by PR #194 which has now been merged. Issue can be closed.

closed 2018-10-18 by sbutcher-arm