ARMmbed/mbed-os-example-tls #54: [OOB_5_4] tls-client got "SSL - Memory allocation failed"

soramame21, 2017-03-02:
I got following error when running tls-client on **NUCLEO_F746ZG**.

[image]

# How to build the bad executable
mbed compile -m NUCLEO_F746ZG -t IAR -c > bb0 2>&1

# How to build a working executable
> mbed compile -m NUCLEO_F746ZG -t GCC_ARM -c > bb2 2>&1

soramame21, 2017-03-07:
@simonb @MarceloSalazar Could you give comment for this issue?

sbutcher-arm, 2017-03-07:
Looks like a memory problem. The suspicion is the heap isn't big enough with IAR.

@mazimkhan - Can you please look at this?

mazimkhan, 2017-03-07:
@soramame21 please try with latest mbed-os that has fix for the memory issue.

soramame21, 2017-03-07:
@mazimkhan I'll try it tomorrow. thanks.

mazimkhan, 2017-03-07:
This PR fixed this issue in mbed-os https://github.com/ARMmbed/mbed-os/pull/3871

mazimkhan, 2017-03-07:
Please try with this PR on mbed-os
https://github.com/ARMmbed/mbed-os/issues/3902

mazimkhan, 2017-03-07:
Updated my comment above. This issue should be fixed by ARMmbed/mbed-os#3902

MarceloSalazar, 2017-03-07:
I confirm PR ARMmbed/mbed-os#3902 fixes memory allocation problem when building with IAR
@0xc0170 please add a note to the mbed OS known issue list

soramame21, 2017-03-08:
@mazimkhan @MarceloSalazar yes, I verified the fix for IAR toolchain as well.

mazimkhan, 2017-03-10:
Can this be closed now?

closed 2017-03-10 by MarceloSalazar