ARMmbed/mbed-os-example-tls #55: [OOB_5_4] tls-client got a compilation error for target HEXIWEAR

soramame21, 2017-03-02:
All of following build commands resulted a compilation error.
## Building commnand
1.
`mbed compile -m HEXIWEAR -t IAR -c > bb0 2>&1`
2.
`mbed compile -m HEXIWEAR -t ARM -c > bb1 2>&1`
3.
`mbed compile -m HEXIWEAR -t GCC_ARM -c > bb2 2>&1`

## Building log

> Building project tls-client (HEXIWEAR, IAR)
> Scan: .
> Scan: FEATURE_BLE
> Scan: FEATURE_COMMON_PAL
> Scan: FEATURE_LWIP
> Scan: FEATURE_UVISOR
> Scan: FEATURE_ETHERNET_HOST
> Scan: FEATURE_LOWPAN_BORDER_ROUTER
> Scan: FEATURE_LOWPAN_HOST
> Scan: FEATURE_LOWPAN_ROUTER
> Scan: FEATURE_NANOSTACK
> Scan: FEATURE_NANOSTACK_FULL
> Scan: FEATURE_THREAD_BORDER_ROUTER
> Scan: FEATURE_THREAD_END_DEVICE
> Scan: FEATURE_THREAD_ROUTER
> Scan: FEATURE_STORAGE
> Scan: mbed
> Scan: env
> Compile [  0.4%]: cmain.S
> Compile [  0.9%]: AnalogIn.cpp
> Compile [  1.3%]: BusIn.cpp
> Compile [  1.7%]: BusInOut.cpp
> Compile [  2.2%]: main.cpp
> [Fatal Error] main.cpp@38,0: [Pe1696]: cannot open source file "EthernetInterface.h"
> [ERROR] 
>   #include "EthernetInterface.h"
>                                 ^
> "C:\iotBu\ooT-228\import\mbed-os-example-tls\tls-client\main.cpp",38  Fatal error[Pe1696]: cannot open source file "EthernetInterface.h"
>             searched: "C:\iotBu\ooT-228\import\mbed-os-example-tls\tls-client\"
>             searched: "./"
>             searched: "./DebugConfig/"
>             searched: "./mbed-os/"
>             searched: "./mbed-os/cmsis/"
>             searched: "./mbed-os/cmsis/TOOLCHAIN_IAR/"
>             searched: "./mbed-os/docs/"
>             searched: "./mbed-os/drivers/"
>             searched: "./mbed-os/events/"
>             searched: "./mbed-os/events/equeue/"
>             searched: "./mbed-os/features/"
>             searched: "./mbed-os/features/frameworks/"
>             searched: "./mbed-os/features/frameworks/greentea-client/"
>             searched: "./mbed-os/features/frameworks/greentea-client/greentea-client/"
>             searched: "./mbed-os/features/frameworks/greentea-client/source/"
>             searched: "./mbed-os/features/frameworks/unity/"
>             searched: "./mbed-os/features/frameworks/unity/source/"
>             searched: "./mbed-os/features/frameworks/unity/unity/"
>             searched: "./mbed-os/features/frameworks/utest/"
>             searched: "./mbed-os/features/frameworks/utest/source/"
>             searched: "./mbed-os/features/frameworks/utest/utest/"
>             searched: "./mbed-os/features/mbedtls/"
>             searched: "./mbed-os/features/mbedtls/importer/"
>             searched: "./mbed-os/features/mbedtls/inc/"
>             searched: "./mbed-os/features/mbedtls/inc/mbedtls/"
>             searched: "./mbed-os/features/mbedtls/platform/"
>             searched: "./mbed-os/features/mbedtls/platform/inc/"
>             searched: "./mbed-os/features/mbedtls/platform/src/"
>             searched: "./mbed-os/features/mbedtls/src/"
>             searched: "./mbed-os/features/nanostack/"
>             searched: "./mbed-os/features/netsocket/"
>             searched: "./mbed-os/features/storage/"
>             searched: "./mbed-os/hal/"
>             searched: "./mbed-os/hal/storage_abstraction/"
>             searched: "./mbed-os/platform/"
>             searched: "./mbed-os/rtos/"
>             searched: "./mbed-os/rtos/rtx/"
>             searched: "./mbed-os/rtos/rtx/TARGET_CORTEX_M/"
>             searched: "./mbed-os/rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/"
>             searched: "./mbed-os/rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_IAR/"
>             searched: "./mbed-os/targets/"
>             searched: "./mbed-os/targets/TARGET_Freescale/"
>             searched: "./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/"
>             searched: "./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/"
>             searched: "./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_HEXIWEAR/"
>             searched: "./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/"
>             searched: "./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_IAR/"
>             searched: "./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/drivers/"
>             searched: "./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/"
>             searched: "./settings/"
>             searched: "C:/iotBu/ooT-228/import/mbed-os-example-tls/tls-client/mbed-os/BUILD"
>             searched: "C:/iotBu/ooT-228/import/mbed-os-example-tls/tls-client/mbed-os/features/unsupported/tests/mbed/"
>             searched: "C:/iotBu/ooT-228/import/mbed-os-example-tls/tls-client/mbed-os/features/unsupported/tests/mbed/env/"
>             searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.5\arm\inc\"
>             searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.5\arm\inc\c\"
>             searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.5\arm\inc\cpp\"
> Fatal error detected, aborting.
> 
> [mbed] ERROR: "python" returned error code 1.
> [mbed] ERROR: Command "python -u C:\iotBu\ooT-228\import\mbed-os-example-tls\tls-client\mbed-os\tools\make.py -t IAR -m HEXIWEAR --source . --build .\BUILD\HEXIWEAR\IAR -c" in "C:\iotBu\ooT-228\import\mbed-os-example-tls\tls-client"
> ---
>

0xc0170, 2017-03-02:
What's the error ? please use markdown code highlighting for console output for better readability. I can't spot the error from the above output.

soramame21, 2017-03-03:
@0xc0170 
I had updated the building log. the point is target HEXIWEAR doesn't have an Ethernet interface. it is not suitable target for tls test.

MarceloSalazar, 2017-03-06:
As HEXIWEAR doesn't have Ethernet, I'd suggest we close this issue.

soramame21, 2017-03-06:
I agree. it's closed now.

closed 2017-03-06 by soramame21