ARMmbed/mbed-os-example-tls #56: [OOB_5_4] tls-client "X509 - Certificate verification failed" using IAR

sunsmilearm, 2017-03-02:
Log is:

```
Using Ethernet LWIP
Client IP Address is 10.169.36.60
Connecting with developer.mbed.org
Starting the TLS handshake...
mbedtls_ssl_handshake() failed: -0x2700 (-9984): X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
MBED: Socket Error: -1
```

0xc0170, 2017-03-02:
Please when reporting an issue, provide as much details as you can. What commands did you use, what target, what toolchain (version as well). Think about how to reproduce this by another user.

sbutcher-arm, 2017-03-02:
@sunsmilearm - Can you please confirm what the board was? This will be a board specific issue.

sunsmilearm, 2017-03-03:
@sbutcher-arm Sorry, forget to mention that the platform is Ublox Odin W2

Tool chain:
IAR C/C++ Compiler for ARM
  7.70.2.11706 (7.70.2.11706)

command is simply mbed compile

sbutcher-arm, 2017-03-07:
This should be fixed in RC2 by https://github.com/ARMmbed/mbed-os/pull/3871

Please retest.

bridadan, 2017-03-07:
I just retested this and everything worked.

mbed-os-example-tls commit: `4cc386327c21917c0d5e1445f2ad1f068ebe24b9`

IAR verison:
```
$ iccarm --version
IAR ANSI C/C++ Compiler V7.80.1.11864/W32 for ARM
```

Log output from device:
```
Using Ethernet LWIP
Client IP Address is 10.118.14.103
Connecting with developer.mbed.org
Starting the TLS handshake...
TLS connection to developer.mbed.org established
Server certificate:
    cert. version     : 3
    serial number     : 11:21:B8:47:9B:21:6C:B1:C6:AF:BC:5D:0C:19:52:DC:D7:C3
    issuer name       : C=BE, O=GlobalSign nv-sa, CN=GlobalSign Organization Validation CA - SHA256 - G2
    subject name      : C=GB, ST=Cambridgeshire, L=Cambridge, O=ARM Ltd, CN=*.mbed.com
    issued  on        : 2016-03-03 12:26:08
    expires on        : 2017-04-05 10:31:02
    signed using      : RSA with SHA-256
    RSA key size      : 2048 bits
    basic constraints : CA=false
    subject alt name  : *.mbed.com, mbed.org, *.mbed.org, mbed.com
    key usage         : Digital Signature, Key Encipherment
    ext key usage     : TLS Web Server Authentication, TLS Web Client Authentication
Certificate verification passed

HTTPS: Received 437 chars from server
HTTPS: Received 200 OK status ... [OK]
HTTPS: Received 'Hello world!' status ... [OK]
HTTPS: Received message:

HTTP/1.1 200 OK
Server: nginx/1.7.10
Date: Tue, 07 Mar 2017 19:27:19 GMT
Content-Type: text/plain
Content-Length: 14
Connection: keep-alive
Last-Modified: Fri, 27 Jul 2012 13:30:34 GMT
Accept-Ranges: bytes
Cache-Control: max-age=36000
Expires: Wed, 08 Mar 2017 05:27:19 GMT
X-Upstream-L3: 172.17.0.4:80
X-Upstream-L2: developer-sjc-cyan-2-nginx
Strict-Transport-Security: max-age=31536000; includeSubdomains

Hello world!
```

sunsmilearm, 2017-03-08:
Fixed. Thanks.

bridadan, 2017-03-08:
@sunsmilearm Could you close this issue then?

closed 2017-03-09 by sunsmilearm