ARMmbed/mbed-os-example-tls #61: [OOB_5_4] tls-client failed when running on NUCLEO_F746ZG

soramame21, 2017-03-06:
tls-client outputted following error message when running on NUCLEO_F746ZG.

[image]

### Imported URL to online compiler
```https://github.com/ARMmbed/mbed-os-example-tls#fcc3b724ff879b7dcfc3a5d4fb5e23b7de5f7848```
then removed other examples (authcrypt/benchmark/hashing), built tls-client only.
I had verified this error happens at master as well.

soramame21, 2017-03-07:
@simonb @MarceloSalazar Could you give comment for this issue?

sbutcher-arm, 2017-03-07:
cc: @andresag01, @bridadan 

I don't believe this is supported by the online IDE. We had a similar bug before in issue https://github.com/ARMmbed/mbed-os/issues/3636.

Can you please retest this by importing the TLS client sample by some other means?

MarceloSalazar, 2017-03-07:
I'm hitting the same run time problem when using mbed CLI with either compiler (ARM, GCC, IAR).
mbed OS RC2: 305f5c491e34d86098e9da91b322017549c189ff
Example OOB branch: 4cc386327c21917c0d5e1445f2ad1f068ebe24b9
Using the Online Compiler shouldn't make a difference (assuming other folders are deleted) and should just work

mazimkhan, 2017-03-07:
As observed together this works with mbed-CLI but fails with online IDE. Something wrong with online IDE environment.

MarceloSalazar, 2017-03-07:
@0xc0170 @sbutcher-arm can we add a (known issue) note to these examples, saying that the Online IDE is not expected to work with this example?

bridadan, 2017-03-07:
As @sbutcher-arm mentioned, this was brought up last time: https://github.com/ARMmbed/mbed-os/issues/3636

The online compiler assumes one project per repo. In face mbed CLI assumes this too when you use `mbed import`, but this repo contains multiple examples and uses a few different tricks to get it to work which don't work in the online compiler. We need to either test the mercurial mirrors on the developer site of each individual example, this repo should be broken apart, or we need to mark this as "DON'T TEST" for the online compiler.

MarceloSalazar, 2017-03-07:
Thanks Brian. I indeed tried to remove other folders to make sure that no multiple main.cpp/mbed-os.lib where in the project. Interestingly this 'trick' works with the BLE examples.
I'd suggest not testing the Online IDE with these examples anymore, as it's not really a valid use case nor great UX.

bridadan, 2017-03-07:
Hmm that is strange. @andresag01 added instructions last time pointing the user to the developer site mirror project: https://github.com/ARMmbed/mbed-os-example-tls/blame/master/authcrypt/README.md#L9

Though perhaps we should make those instructions more prominent? They're kind of easy to miss right now.

soramame21, 2017-03-08:
@bridadan @MarceloSalazar Why is revision of **tls-client\mbed-os** not updated at online IDE? but revision of **tls-clinet** can be updated!
### Revision of `tls-client` is updated 
[image]
### Revision of `tls-client\mbed-os` is not updated 
[image]

MarceloSalazar, 2017-03-08:
> Why is revision of tls-client\mbed-os not updated at online IDE? 
This is a limitation of the (beta) Online IDE when working with mbed OS versions that are not official (e.g. RC2)

soramame21, 2017-03-08:
I got it, Thanks.

MarceloSalazar, 2017-03-08:
The doc shows that we should be using the developer.mbed.org examples for the Online IDE:
https://github.com/ARMmbed/mbed-os-example-tls#required-software
This is a known limitation of the Online IDE.

@0xc0170

closed 2017-03-08 by MarceloSalazar