ARMmbed/mbed-os-example-tls #65: Cannot export to IAR, platform: UBLOX_EVK_ODIN_W2

sunsmilearm, 2017-03-14:
command:
  cd authcrypt
  mbed compile -i iar -m UBLOX_EVK_ODIN_W2

error log:

```
Scan: .
Traceback (most recent call last):
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project.py", line 248, in <module>
    main()
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project.py", line 244, in main
    build_profile=profile)
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project.py", line 94, in export
    build_profile=build_profile, silent=silent)
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project_api.py", line 222, in export_project
    macros=macros)
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project_api.py", line 89, in generate_project_files
    exporter.generate()
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\export\iar\__init__.py", line 122, in generate
    'linker_script': self.format_file(self.resources.linker_script),
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\export\iar\__init__.py", line 89, in format_file
    return join('$PROJ_DIR$',file)
  File "c:\python27\lib\ntpath.py", line 67, in join
    p_drive, p_path = splitdrive(p)
  File "c:\python27\lib\ntpath.py", line 115, in splitdrive
    if len(p) > 1:
TypeError: object of type 'NoneType' has no len()
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project.py -i iar -m UBLOX_EVK_ODIN_W2 --source ." in "C:\my\dev\mbed_5_0\oob_540\tls_n\au
thcrypt"
---
```

andresag01, 2017-03-14:
@theotherjimmy: Could you please take a look at this issue with the exporter?

ashok-rao, 2017-03-15:
@theotherjimmy , can you please update any status on this? Thanks!

andresag01, 2017-06-06:
I think the problem here is that the mbed-cli command used is incorrect. To compile with IAR it should be `-t` instead of `-i`. If the intention is to export, then the command should be something like `mbed export -i make_iar -m UBLOX_EVK_ODIN_W2`. Closing this issue.

closed 2017-06-06 by andresag01