ARMmbed/mbed-os-example-tls #66: cannot export to make_iar, platform UBLOX_EVK_ODIN_W2

sunsmilearm, 2017-03-14:
command:
cd authcrypt
mbed compile -i make_iar -m UBLOX_EVK_ODIN_W2

log:
```
Scan: .
Traceback (most recent call last):
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project.py", line 248, in <module>
    main()
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project.py", line 244, in main
    build_profile=profile)
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project.py", line 94, in export
    build_profile=build_profile, silent=silent)
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project_api.py", line 222, in export_project
    macros=macros)
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project_api.py", line 89, in generate_project_files
    exporter.generate()
  File "C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\export\makefile\__init__.py", line 49, in generate
    raise NotSupportedException("No linker script found.")
tools.utils.NotSupportedException: No linker script found.
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\my\dev\mbed_5_0\oob_540\tls_n\authcrypt\mbed-os\tools\project.py -i make_iar -m UBLOX_EVK_ODIN_W2 --source ." in "C:\my\dev\mbed_5_0\oob_540\tls
_n\authcrypt"
---
```

andresag01, 2017-03-14:
@theotherjimmy: Could you please take a look at this issue with the exporter?

ashok-rao, 2017-03-15:
@theotherjimmy , bump!

theotherjimmy, 2017-03-15:
@sunsmilearm I think your command invocation is incorrect
> `mbed compile ...`

should be `mbed export ...`

I have seen a few issues related to exporting for the odin. Further, I cannot reproduce this issue.

Versions:
 - mbed - 1.0.0
 - git show (in authcrypt) 
```
commit f76af378cd3a506b68867529422b8cbce897d7c6 (HEAD -> master, tag: mbed-os-5.4.0, origin/master, origin/HEAD)
Merge: 20d415a d88d5a0
Author: Andres Amaya Garcia <andres.amayagarcia@arm.com>
Date:   Thu Mar 9 14:10:03 2017 +0000

    Merge pull request #64 from andresag01/oob_test_mbed-os-5.4
    
    Updating mbed-os to mbed-os-5.4.0
```
 - git show (in mbed-os)
```
commit 305f5c491e34d86098e9da91b322017549c189ff (HEAD, tag: mbed-os-5.4.0-rc2, tag: mbed-os-5.4.0, origin/mbed-os-5.4, mbed-os-5.4)
Merge: 3a27568a5 57b4d80e8
Author: Martin Kojtal <0xc0170@gmail.com>
Date:   Sun Mar 5 18:03:02 2017 +0000

    Merge pull request #3882 from ARMmbed/release-candidate
    
    Release candidate for mbed-os-5.4.0-rc2
```

andresag01, 2017-06-06:
I have tested this issue with mbed-os-5.4.6 and the problem seems to have been resolved. Closing the issue.

closed 2017-06-06 by andresag01