ARMmbed/mbed-os-example-tls #67: Cannot export to uvision5, platform UBLOX_EVK_ODIN_W2

sunsmilearm, 2017-03-14:
commands:
cd benchmark
mbed export -i uvision5 -m UBLOX_EVK_ODIN_W2

the export log is only "scan ." and the project is nearly empty.

andresag01, 2017-03-14:
@theotherjimmy: Is this the expected behaviour? I ran the command and I get the following output:
```

ashok-rao, 2017-03-14:
@andresag01 .. The exported project does not contain anything except 1 file (mbed_config.h)...

andresag01, 2017-03-14:
@ashok-rao: That is interesting. I ran the same test and I got the files that I listed above. Perhaps it would be useful if you can post some additional information regarding your environment to help reproduce the issue. For example, what mbed-cli version are you using? what OS are you running? etc.

ashok-rao, 2017-03-14:
@andresag01 , when I run export, I get the *.uvprojx & *.uvoptx (which are the project files for uVision). When I open the project with MDK, there is nothing ..

[image]

andresag01, 2017-03-14:
@ashok-rao: Thanks for the additional information. Perhaps @theotherjimmy can help us with this problem?

theotherjimmy, 2017-03-14:
again, https://github.com/ARMmbed/mbed-os/pull/3924 is the solution.

mehdiBezahaf, 2017-03-29:
Had the same problem and ARMmbed/mbed-os#3924 fixed it ;) Thanks

andresag01, 2017-06-07:
I have tried this with mbed OS 5.4.6 and the issue seems to be fixed. Closing this ticket...

closed 2017-06-07 by andresag01