ARMmbed/mbed-os-example-tls #68: [OoB_5.4.1] [EXPORTER] : export to uvision5 and make_armc5 fails on NuMaker_PFM_NUC472.

ashok-rao, 2017-03-14:
**Steps to reproduce:**
1. import example, update to oob-test-mbed-os-5.4.1 branch.
2. mbed export -i uvision5 -m NUMAKER_PFM_NUC472

**Result:**
Scan: .
(end)

**Exporting with -vv option:**
[mbed] Working path "C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt" (library)
[mbed] Exec "python -u C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt\mbed-os\tools\project.py -i uvision5 -m NUMAKER_PFM_NUC472 --source ." in C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt
Scan: .
(end)

Problem with exporter. Raising here to keep track.

ashok-rao, 2017-03-14:
For export to make_armc5:
```
C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt>mbed export -i make_armc5 -m NUMAKER_PFM_NUC472 -vv
[mbed] Working path "C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt" (library)
[mbed] Exec "python -u C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt\mbed-os\tools\project.py -i make_armc5 -m NUMAKER_PFM_NUC472 --source ." in C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt
Scan: .
Traceback (most recent call last):
  File "C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt\mbed-os\tools\project.py", line 248, in <module>
    main()
  File "C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt\mbed-os\tools\project.py", line 244, in main
    build_profile=profile)
  File "C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt\mbed-os\tools\project.py", line 94, in export
    build_profile=build_profile, silent=silent)
  File "C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt\mbed-os\tools\project_api.py", line 222, in export_project
    macros=macros)
  File "C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt\mbed-os\tools\project_api.py", line 89, in generate_project_files
    exporter.generate()
  File "C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt\mbed-os\tools\export\makefile\__init__.py", line 49, in generate
    raise NotSupportedException("No linker script found.")
tools.utils.NotSupportedException: No linker script found.
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt\mbed-os\tools\project.py -i make_armc5 -m NUMAKER_PFM_NUC472 --source ." in "C:\Ashok\mbed_CLI\v5_4_OoB\v_5_4_1_OoB\mbed-os-example-tls\authcrypt"
---
```

ashok-rao, 2017-03-14:
Issue also seen with "benchmark" example exporting to "make_gcc_arm".

andresag01, 2017-03-14:
@theotherjimmy: Could you please take a look at this issue with the exporter?

theotherjimmy, 2017-03-14:
That's fixed by https://github.com/ARMmbed/mbed-os/pull/3924

MarceloSalazar, 2017-03-17:
Issue ARMmbed/mbed-os#3924 still open. Waiting to be merged and confirm fix.

ashok-rao, 2017-03-30:
https://github.com/ARMmbed/mbed-os/pull/3924 merged and released for 5.4.2. Closing issue!

closed 2017-03-30 by ashok-rao