ARMmbed/mbed-os-example-tls #69: wrong output in the Benchmark application

RonEld, 2017-03-16:
The output of the Benchmark Application performance is in Kb/s , however, according to the code:
~~~
    mbedtls_printf( "%9lu Kb/s,  %9lu cycles/byte\r\n",                        \
                     i * BUFSIZE / 1024,                                       \
                     ( mbedtls_timing_hardclock() - tsc ) / ( j * BUFSIZE ) );
~~~
The performance should be in Kilo Bytes (KB) not Kilo bits (Kb) as printed

andresag01, 2017-06-07:
A fix for this issue has already been merged into master. Closing ticket...

closed 2017-06-07 by andresag01

RonEld, 2017-06-07:
@andresag01 I think this issue was remained opened because of the benchmark application in the mbed TLS repository: https://github.com/ARMmbed/mbedtls/pull/920

RonEld, 2017-06-07:
my mistakes, I thought about the internal JIRA ticket