ARMmbed/mbed-os-example-tls #91: Authcrypt compiled with make_gcc_arm fails with runtime error

bridadan, 2017-06-08:
With branch `mbed-os-5.5.0-rc1-oob`, when exporting with `make_gcc_arm`, the example fails to run.

I received two different errors when running the authcrypt example.

```
plaintext message: 536f6d65207468696e67732061726520626574746572206c65667420756e7265616400
CMSIS-RTOS error: Stack underflow (status: 0x1, task ID: 0x20006CE8, task name: (null))
```

```
plaintext message: 536f6d65207468696e67732061726520626574746572206c65667420756e7265616400
Mutex 0x730fcfdb error -4: Parameter error
```

deepikabhavnani, 2017-06-09:
@bridadan on which target device?

yennster, 2017-06-09:
I received a similar error on HEXIWEAR with the following commands: `mbed export -i make_gcc_arm -m HEXIWEAR` & `make` -

```
plaintext message: 536f6d65207468696e67732061726520626574746572206c65667420756e7265616400
CMSIS-RTOS error: Stack underflow (status: 0x1, task ID: 0x20003C00, task name: (null))
```

yennster, 2017-06-09:
@deepikabhavnani @bridadan Once again, I received a similar error on K66F with the following commands: `mbed export -i make_gcc_arm -m K66F` & `make` -

```
plaintext message: 536f6d65207468696e67732061726520626574746572206c65667420756e7265616400
CMSIS-RTOS error: Stack underflow (status: 0x1, task ID: 0x20003400, task name: (null))
```

andresag01, 2017-06-12:
I tried with both HEXIWEAR and K64F and the example fails in both cases when the application is compiled with the debug profile. For example, compiling with the following command will result in the same stack underflow error:
```
mbed compile -t GCC_ARM -m K64F --profile mbed-os/tools/profiles/debug.json
```

bridadan, 2017-06-12:
Sorry y'all, it was late on a Thursday, wasn't thinking straight. It was on the UBLOX_EVK_ODIN_W2.

@andresag01 good catch about the debug profile, maybe this is a generic issue and not just for the TLS examples...

andresag01, 2017-06-13:
It seems that we are running out of stack memory and a runtime check seems to be throwing this failure. As I pointed out before, the problem only seems to happen with debug builds, while no error occurs with release or develop builds probably because in debug builds compiler optimisations are completely disabled.

This ticket is related to https://github.com/ARMmbed/mbed-os-example-tls/issues/14.

bridadan, 2017-06-13:
@andresag01 Perhaps you can look at moving some of the memory that's allocated on the main thread to global space?

closed 2017-07-14 by andresag01