ARMmbed/mbed-os-example-tls #92: Benchmark example built with make_iar thows runtime error

bridadan, 2017-06-08:
Once the benchmark example finishes running, a CMSIS RTOS error is thrown when built with the `make_iar` exporter. Here's the full log:

```
MD4                      :      12048 Kb/s,         13 cycles/byte
MD5                      :       9120 Kb/s,         17 cycles/byte
SHA-1                    :       3973 Kb/s,         41 cycles/byte
SHA-256                  :       1135 Kb/s,        144 cycles/byte
SHA-512                  :        602 Kb/s,        273 cycles/byte
ARC4                     :       5272 Kb/s,         31 cycles/byte
3DES                     :        215 Kb/s,        767 cycles/byte
DES                      :        608 Kb/s,        270 cycles/byte
AES-CBC-128              :       1455 Kb/s,        112 cycles/byte
AES-CBC-192              :       1262 Kb/s,        130 cycles/byte
AES-CBC-256              :       1114 Kb/s,        147 cycles/byte
AES-GCM-128              :        635 Kb/s,        259 cycles/byte
AES-GCM-192              :        595 Kb/s,        276 cycles/byte
AES-GCM-256              :        560 Kb/s,        293 cycles/byte
AES-CCM-128              :        600 Kb/s,        274 cycles/byte
AES-CCM-192              :        532 Kb/s,        309 cycles/byte
AES-CCM-256              :        477 Kb/s,        344 cycles/byte
CTR_DRBG (NOPR)          :       1133 Kb/s,        144 cycles/byte
CTR_DRBG (PR)            :        813 Kb/s,        202 cycles/byte
HMAC_DRBG SHA-1 (NOPR)   :        275 Kb/s,        600 cycles/byte
HMAC_DRBG SHA-1 (PR)     :        255 Kb/s,        648 cycles/byte
HMAC_DRBG SHA-256 (NOPR) :        137 Kb/s,       1209 cycles/byte
HMAC_DRBG SHA-256 (PR)   :        121 Kb/s,       1373 cycles/byte
RSA-2048                 :      38 ms/ public
RSA-2048                 :    1504 ms/private
RSA-4096                 :     134 ms/ public
RSA-4096                 :    8617 ms/private
ECDHE-secp384r1          :     951 ms/handshake
ECDHE-secp256r1          :     610 ms/handshake
ECDHE-Curve25519         :     514 ms/handshake
ECDH-secp384r1           :     467 ms/handshake
ECDH-secp256r1           :     298 ms/handshake
ECDH-Curve25519          :     264 ms/handshake

DONE
CMSIS-RTOS error: Stack underflow (status: 0x1, task ID: 0x2001C2F4, task name: hd)
```

andresag01, 2017-06-09:
@bridadan: Could you please tell us which target you were using?

bridadan, 2017-06-12:
Oh jeez, that would be helpful wouldn't it? *facepalm*

It was the UBLOX_EVK_ODIN_W2.

andresag01, 2017-06-13:
It seems that we are running out of stack memory and a runtime check seems to be throwing this failure. As I pointed out before, the problem only seems to happen with debug builds, while no error occurs with release or develop builds probably because in debug builds compiler optimisations are completely disabled.

This ticket is related to https://github.com/ARMmbed/mbed-os-example-tls/issues/14.

RonEld, 2018-07-26:
I believe that #116 may have resolved this

andresag01, 2018-10-22:
As @RonEld pointed out, this problem was solved in #116 and #194, so this ticket can now be closed. Please reopen if you think that the problem persists or further work is needed.

closed 2018-10-22 by andresag01