ARMmbed/mbed-os-example-tls #98: No server example for mbedTLS on mbed

sherrellbc, 2017-06-29:
Is such a thing possible on mbed-os? I found [this](https://github.com/ARMmbed/mbedtls/blob/development/programs/ssl/ssl_server.c) example on the mbedTLS Github page, but the code was written assuming a UNIX platform. 

After searching around a bit I have found examples:
- [tls.mbed](https://tls.mbed.org/discussions/generic/compiling-problems) 
- [developer.mbed](https://developer.mbed.org/questions/69356/How-the-communication-between-client-ser/)
- [developer.mbed](https://developer.mbed.org/questions/76079/mbedtls-linker-error/)

All pointing to there not being any server support for mbedTLS on mbed-os. If this is not sure, to what extent has server support been implemented in mbed's port of mbedTLS?

sherrellbc, 2017-06-29:
I just posted over at [mbed](https://github.com/ARMmbed/mbed-os/issues/4670) as it's more appropriately the place to be asking about support on the mbed platform. However, I'll leave this here since any resulting examples are likely to be published to this repo.

andresag01, 2017-06-29:
@sherrellbc: The conversations that you have linked in this issue do not state that there is not mbed TLS support for servers in mbed OS. In fact, the module that is mentioned there as being "UNIX/Windows only" is `net_sockets.c`, but this in no way impairs the user of mbed TLS as a server in mbed OS.

The mbed TLS `net_sockets.c` module is simply wrapper code around the UNIX/Windows sockets API. In the mbed TLS examples [here](https://github.com/ARMmbed/mbedtls/tree/development/programs/ssl) function pointers to the `net_sockets.c` functionality is configured into the mbed TLS SSL context. However mbed OS has different socket APIs, so `net_sockets.c` cannot be used directly. Instead, you need to provide your own wrappers to the TLS stack as demonstrated in [this](https://github.com/ARMmbed/mbed-os-example-tls/blob/master/tls-client/main.cpp#L199) mbed OS example.

I hope this helps.

andresag01, 2017-09-11:
The question has been answered and there has not been any activity in this issue for sometime. Therefore, I am closing this issue. Please feel free to reopen it if you have further queries.

closed 2017-09-11 by andresag01