ARMmbed/mbed-os-example-tls #100: Integrate with easy-connect

JanneKiiskila, 2017-07-05:
In order to enable more network bearers w/o hacking the main.cpp OR doing many branches for it, should we use the easy-connect instead?

@janjongboom

JanneKiiskila, 2017-07-07:
PR proposal in; https://github.com/ARMmbed/mbed-os-example-tls/pull/101

janjongboom, 2017-07-12:
I like it.

andresag01, 2017-10-17:
@JanneKiiskila: Thanks for raising this. We have started reviewing your PR. Also, please note that we had already considered adding support for other transports in #43.

janjongboom, 2017-10-20:
Note that I have https://developer.mbed.org/teams/sandbox/code/http-example, which uses easy-connect already and does talk HTTPS. Maybe we should use that in favor of tls-client.

andresag01, 2017-11-02:
PR https://github.com/ARMmbed/mbed-os-example-tls/pull/101 has already been merged into the tls-client example. Closing this ticket...

With regards to @janjongboom, the tls-client is designed to be a minimal example of how to use Mbed TLS to setup a secure connection and is also used for other purposes such as out-of-the-box testing. The http-example you linked seems a lot more complex. Perhaps dropping tls-client would be something interesting to consider in the future, but this would have to be discussed.

@sbutcher-arm any thoughts?

closed 2017-11-02 by andresag01