ARMmbed/mbed-os-example-tls #122: Why 9600 baud, if you already comment in main it might timeout?

JanneKiiskila, 2017-10-10:
Hard to see why one should stick with 9600 bauds, if 115200 will work reliably with any board I'm aware of.

andresag01, 2017-10-17:
@JanneKiiskila: The baud rate for all mbed TLS example applications has been left to the default mbed OS, which I believe is 9600. In fact, none of our examples explicitly set the baud rate in the code or the .json config files. Given that these are meant to be easy-to-use examples, I think it makes sense to be consistent with the default rate to reduce the chance of new users misconfiguring the serial terminal. However, as you previously mentioned, there is also a case for using a higher baud rate, e.g. 115200.

@sbutcher-arm: What do you think?

sbutcher-arm, 2017-10-17:
Hi @JanneKiiskila. Baud rates in Mbed OS examples have been the subject of passionate debates before, and in my view is an unnecessary distraction. If you're not carrying much data, and it's just debug output for users at a terminal, 9600 is as good as 115200.

In my view we should *only* change the baud rate if all other examples change at the same time.

If you want all the examples to change, please raise the issue with @MarceloSalazar and @sg-.

andresag01, 2017-11-02:
@sbutcher-arm: Thanks for the input.

@JanneKiiskila: Thank you for raising this ticket nonetheless. Given @sbutcher-arm's comment, I will close this issue and if a decision is reached to change the baud rate for these examples I will reopen it. I hope this is OK with you.

closed 2017-11-02 by andresag01