ARMmbed/mbed-os-example-tls #133: Benchmark does not compile with IAR + Odin

sarahmarshy, 2017-12-11:
Here is the compilation error:
```
mbed compile -t iar -m auto
[mbed] Detected "UBLOX_EVK_ODIN_W2" connected to "D:" and using com port "COM5"
Building project mbed-os-example-tls (UBLOX_EVK_ODIN_W2, IAR)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Compile [  0.5%]: AnalogIn.cpp
[Error] mpu_armv7.h@167,21: [Pe018]: expected a ")"
[Error] mpu_armv7.h@170,15: [Pe020]: identifier "len" is undefined
[Error] mpu_armv7.h@172,47: [Pe020]: identifier "src" is undefined
[Error] mpu_armv7.h@184,47: [Pe140]: too many arguments in function call
[Error] mpu_armv7.h@187,0: [Pe140]: too many arguments in function call
[ERROR]
  __STATIC_INLINE void orderedCpy(volatile uint32_t* dst, const uint32_t* __RESTRICT src, uint32_t len)
                                                                                     ^
"C:\Repos\mbed-os-example-tls\benchmark\mbed-os\cmsis\TARGET_CORTEX_M\mpu_armv7.h",167  Error[Pe018]: expected a ")"

    for (i = 0U; i < len; ++i)
                     ^
"C:\Repos\mbed-os-example-tls\benchmark\mbed-os\cmsis\TARGET_CORTEX_M\mpu_armv7.h",170  Error[Pe020]: identifier "len" is undefined

      dst[i] = src[i];
               ^
"C:\Repos\mbed-os-example-tls\benchmark\mbed-os\cmsis\TARGET_CORTEX_M\mpu_armv7.h",172  Error[Pe020]: identifier "src" is undefined

      orderedCpy(&(MPU->RBAR), &(table->RBAR), MPU_TYPE_RALIASES*rowWordSize);
                                               ^
"C:\Repos\mbed-os-example-tls\benchmark\mbed-os\cmsis\TARGET_CORTEX_M\mpu_armv7.h",184  Error[Pe140]: too many arguments in function call

      orderedCpy(&(MPU->RBAR), &(table->RBAR), cnt*rowWordSize);
                                               ^
"C:\Repos\mbed-os-example-tls\benchmark\mbed-os\cmsis\TARGET_CORTEX_M\mpu_armv7.h",187  Error[Pe140]: too many arguments in function call

[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Repos\mbed-os-example-tls\.temp\tools\make.py -t iar -m UBLOX_EVK_ODIN_W2 --source . --build .\BUILD\UBLOX_EVK_ODIN_W2\iar" in "C:\Repos\mbed-os-example-tls"
---
```

sbutcher-arm, 2017-12-11:
@0xc0170 - This looks like an issue in CMSIS - `cmsis/TARGET_CORTEX_M/mpu_armv7.h`. Can you give guidance as to where I should direct this issue?

0xc0170, 2017-12-12:
mbed-os, it seems like this is cmsis file comes from https://github.com/ARM-software/CMSIS_5/blob/04554249ab16ca2df1bb0b2562552bd7286cbfb7/CMSIS/Core/Include/mpu_armv7.h . Please report this to mbed-os, we will investigate with CMSIS team if needed.

adbridge, 2017-12-13:
@sbutcher-arm @mazimkhan Can we get an issue filed against mbed-os for this please and linked here.

mazimkhan, 2017-12-13:
Interesting. With IAR8 I have different set of errors in CI https://jenkins-internal.mbed.com/job/mbedtls-mbed-os-examples-pr/73/execution/node/903/log/
```
11:43:36 [Error] stm32f4xx_hal_def.h@132,30: [Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int const volatile *"
11:43:36 [Error] stm32f4xx_hal_def.h@134,34: [Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int volatile *"
11:43:36 [Error] stm32f4xx_hal_def.h@142,30: [Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int const volatile *"
11:43:36 [Error] stm32f4xx_hal_def.h@144,0: [Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int volatile *"
11:43:36 [ERROR] 
11:43:36   		newValue = (uint32_t)__LDREXW((volatile unsigned long *)ptr) | mask;
11:43:36   		                              ^
11:43:36 "C:\builds\ws\mbedtls-mbed-os-examples-pr@2\benchmark\mbed-os\targets\TARGET_STM\TARGET_STM32F4\device\stm32f4xx_hal_def.h",132  Error[Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int const volatile *"
11:43:36 
11:43:36   	} while (__STREXW(newValue,(volatile unsigned long*) ptr));
11:43:36   	                           ^
11:43:36 "C:\builds\ws\mbedtls-mbed-os-examples-pr@2\benchmark\mbed-os\targets\TARGET_STM\TARGET_STM32F4\device\stm32f4xx_hal_def.h",134  Error[Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int volatile *"
11:43:36 
11:43:36   		newValue = (uint32_t)__LDREXW((volatile unsigned long *)ptr) &~mask;
11:43:36   		                              ^
11:43:36 "C:\builds\ws\mbedtls-mbed-os-examples-pr@2\benchmark\mbed-os\targets\TARGET_STM\TARGET_STM32F4\device\stm32f4xx_hal_def.h",142  Error[Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int const volatile *"
11:43:36 
11:43:36   	} while (__STREXW(newValue,(volatile unsigned long*) ptr));
11:43:36   	                           ^
11:43:36 "C:\builds\ws\mbedtls-mbed-os-examples-pr@2\benchmark\mbed-os\targets\TARGET_STM\TARGET_STM32F4\device\stm32f4xx_hal_def.h",144  Error[Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int volatile *"
11:43:36 
11:43:36 [mbed] ERROR: "python" returned error code 1.
11:43:36 [mbed] ERROR: Command "python -u C:\builds\ws\mbedtls-mbed-os-examples-pr@2\benchmark\mbed-os\tools\make.py -t IAR -m UBLOX_EVK_ODIN_W2 --source . --build .\BUILD\UBLOX_EVK_ODIN_W2\IAR" in "C:\builds\ws\mbedtls-mbed-os-examples-pr@2\benchmark"
11:43:36 ---
```

MarceloSalazar, 2017-12-14:
@mazimkhan we support IAR 7.8 in Mbed OS 5.7
Can we confirm the issue is still valid with this version of the compiler?

mazimkhan, 2017-12-14:
Ok sure I will switch back to 7.8 for Mbed OS 5.7.

mazimkhan, 2017-12-14:
> @sbutcher-arm @mazimkhan Can we get an issue filed against mbed-os for this please and linked here.

Done https://github.com/ARMmbed/mbed-os/issues/5706

MarceloSalazar, 2017-12-18:
@sarahmarshy could you please re-test and confirm? thanks

andreaslarssonublox, 2018-01-09:
I also tried to compile this on mbed-os-example-tls SHA-1: 785f2eaf38edc806cdbe3548247e442a31920ce6
mbed-os 5.7 tag SHA-1: eca67ca7dafab4ef70c21e2463b541132d0dd691

I didn't get the errors as listen in https://github.com/ARMmbed/mbed-os-example-tls/issues/133 but it succeeded. I used IAR WB 7.5.

quantum-leaps, 2018-01-16:
The issue is caused by compiling the C header file "mpu_armv7.h" with the IAR C++ compiler. Specifically, unlike the IAR C, the IAR C++ does NOT recognize the "restrict" keyword by default. The workaround is to provide an explicit command-line option --enable_restrict to the compiler. In the IAR EWARM IDE, open the "Options..." dialog box for your project, choose the C/C++ Category on the left, Extra Options tab. In this tab, check "Use command line options" and add "--enable_restrict" line to the edit control.

sohaibqamar, 2018-08-08:
Hi,
I'm using IAR Workbench 8.30.1 to Build and Debug the code for mbed NXP LPC 1768. 
I have installed only the IAR for ARM and exported the mbed-blinky from online compiler.
I'm trying to build with default options of IAR. I am getting the following errors during Make. Please help.

`Building configuration: mbed_blinky - mbed_blinky 
Updating build tree... 
main.cpp  
Linking 
Error[Li060]: module "mbed_retarget.o" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library. 
Error[Li009]: runtime model conflict: Module mbed_retarget.o specifies that '__CPP_Language' must be 'C++', but module main.o has the value 'C++14'. Modules compiled with C++14 are not link-compatible with modules that are compiled  
with older C++. 
Error[Li060]: module "mbed_poll.o(mbed.a)" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library. 
Error[Li060]: module "FilePath.o(mbed.a)" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library. 
Error[Li060]: module "FileHandle.o(mbed.a)" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library. 
Error[Li060]: module "Timer.o(mbed.a)" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library. 
Error[Li060]: module "FileBase.o(mbed.a)" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library. 
Error[Li005]: no definition for "__aeabi_read_tp" [referenced from errno.o(dl7M_tlf.a)] 
Error while running Linker 
 
Total number of errors: 8 
Total number of warnings: 1 
`

cmonr, 2018-08-08:
@sohaibqamar I've deleted your comment because it an obvious case of thread-jacking, and have moved it into it's own issue over at https://github.com/ARMmbed/mbed-os/issues/7733

RonEld, 2018-08-27:
I have compiled the benchmark application for both release and debug profiles, using
~~~
mbed compile -t IAR -m UBLOX_EVK_ODIN_W2 
~~~
and couldn't reproduce the build error. I have used commit id 714adc8a31e267aa8fbe9a83e689612d487003f9 for the examples repository. 
The mbed-os version is 5.9

Does this issue still reproduce?
i suggest closing this issue as both it doesn't reproduce and it's an Mbed-OS issue, and opened there as well.

screamerbg, 2018-09-19:
Can we close the issue as it's no longer reproducible? @sarahmarshy

closed 2018-10-10 by sarahmarshy

darboy, 2020-05-25:
> The issue is caused by compiling the C header file "mpu_armv7.h" with the IAR C++ compiler. Specifically, unlike the IAR C, the IAR C++ does NOT recognize the "restrict" keyword by default. The [...]

This method is effective ,thank u