ARMmbed/mbed-os-example-tls #134: [Docs Request] .mbedignore or preferred method for selecting examples

sarahmarshy, 2017-12-11:
Please document how to compile only the desired example. I used `.mbedignore` to ignore the examples I did not want to compile. Unfortunately, this means that the mbed-os tools are not at the root of the project, which means that the mbed 2 SDK tools are used. 

The [current suggestion for building](https://github.com/ARMmbed/mbed-os-example-tls/tree/mbed-os-5.7.0-oob#building-and-running-the-examples) would fail because there are multiple copies of mbed-os inside the examples.

Please document the use of `.mbedignore` or other preferred method.

mazimkhan, 2017-12-11:
Can you please specify in the documentation at what step it fails. If you do ```mbed deploy``` from the example sub directory (step 2) then multiple mbed-os.lib files should not interfere.

sarahmarshy, 2017-12-11:
Ah, I ran mbed import instead of git clone for step one.

mazimkhan, 2017-12-12:
> Still, if you want to compile another example, then you must delete the deployed code.
In this repo every example is in a separate directory. Manually we just change directory and deploy->build->test the example. I am not sure how ```mbed import``` works. Does it not support multiple examples repo like this one.

mazimkhan, 2017-12-12:
Just tried ```mbed import``` and it correctly fetched mbed-os in each example dir. I am able to build each example separately. I don't think you need to delete what ```mbed import``` downloaded to switch to a different example. 
The only problem is that this repo isn't compatible with ```mbed import``` command. That expects a repo to be a single mbed application with mbed-os.lib at root. Instead it is a collection of programs. I don't know how much it will help by changing the instructions/build system/mbed-cli for this repo. Sticking to the instructions might be the best way.

```
mazimkhan@mazimkhan-VirtualBox:~/github/ARMmbed$ mbed import https://github.com/ARMmbed/mbed-os-example-tls
[mbed] Importing program "mbed-os-example-tls" from "https://github.com/ARMmbed/mbed-os-example-tls" at latest revision in the current branch
[mbed] Adding library "authcrypt/mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #5f6572179d66

[mbed] Adding library "benchmark/mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #5f6572179d66
[mbed] Adding library "tls-client/easy-connect" from "https://github.com/ARMmbed/easy-connect" at rev #e44b96188010
[mbed] Adding library "tls-client/easy-connect/ATParser" from "https://github.com/ARMmbed/ATParser" at rev #0a74c97ec366
[mbed] Adding library "tls-client/easy-connect/stm-spirit1-rf-driver" from "https://github.com/ARMmbed/stm-spirit1-rf-driver" at rev #0ff4ca7537f0
[mbed] Adding library "tls-client/easy-connect/wifi-x-nucleo-idw01m1" from "https://github.com/ARMmbed/wifi-x-nucleo-idw01m1" at rev #5831f72291b7
[mbed] Adding library "tls-client/easy-connect/wifi-x-nucleo-idw01m1/ATParser" from "https://github.com/ARMmbed/ATParser" at latest revision in the current branch
[mbed] Adding library "tls-client/easy-connect/mcr20a-rf-driver" from "https://github.com/ARMmbed/mcr20a-rf-driver" at rev #9aac10474702
[mbed] Adding library "tls-client/easy-connect/esp8266-driver" from "https://github.com/ARMmbed/esp8266-driver" at rev #fdb44a4bedbb
[mbed] Adding library "tls-client/easy-connect/esp8266-driver/ESP8266/ATParser" from "https://github.com/ARMmbed/ATParser" at rev #a2cb09990f11
[mbed] Adding library "tls-client/easy-connect/atmel-rf-driver" from "https://github.com/ARMmbed/atmel-rf-driver" at rev #3cc6010188c0
[mbed] Adding library "tls-client/mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #5f6572179d66
[mbed] Adding library "hashing/mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #5f6572179d66
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
[mbed] ERROR: Could not execute "hg".
[mbed] ERROR: Please verify that it's installed and accessible from your current path by executing "hg".
---

```

adbridge, 2017-12-13:
@sarahmarshy This repo has been structured this way for as long as i can remember. I am surprised we are only seeing an issue now. @MarceloSalazar How have we tested this previously? Have we just used the mirrored one on os.mbed.com ?

MarceloSalazar, 2017-12-13:
We've been testing this as per instructions shown here: https://github.com/ARMmbed/mbed-os-example-tls#building-and-running-the-examples
(Using git clone / mbed deploy)

This is ok for now (5.7 release) but we'll be raising a requirement to split examples apps to their own repository and improve UX, as the current solution is not good enough.

MarceloSalazar, 2017-12-13:
IOTPREQ-2425

mazimkhan, 2017-12-20:
Closing as nothing to do at the moment. Will be worked upon in IOTPREQ-2425.

closed 2017-12-20 by mazimkhan