ARMmbed/mbed-os-example-tls #137: authcrypt example fails to link on IAR 8.2 + Ublox ODIN

mazimkhan, 2017-12-14:
authcrypt example fails to link with IAR8.2+Ublox ODIN. error:

log https://jenkins-internal.mbed.com/job/mbedtls-mbed-os-examples-pr/74/execution/node/828/log/
```
14:56:57 Link: authcrypt
14:56:57 Error[Li060]: module "cb_main.o(libublox-odin-w2-driver.a)" contains C++ code
14:56:57           built for use with the (E)C++98 library. This code is not link
14:56:57           compatible with the new (C++14) library.
14:56:57 Error[Li009]: runtime model conflict: Module
14:56:57           cb_main.o(libublox-odin-w2-driver.a) specifies that '__CPP_Language'
14:56:57           must be 'C++', but module authcrypt.o has the value 'C++14'. Modules
14:56:57           compiled with C++14 are not link-compatible with modules that are
14:56:57           compiled with older C++.
14:56:57 Error[Li060]: module "cb_ticker_wrapper.o(libublox-odin-w2-driver.a)" contains
14:56:57           C++ code built for use with the (E)C++98 library. This code is not
14:56:57           link compatible with the new (C++14) library.
14:56:57 Warning[Lt009]: Inconsistent wchar_t size
14:56:57             cb_main.o(libublox-odin-w2-driver.a) and 106 other objects have
14:56:57                       wchar_t size 16 bits
14:56:57             authcrypt.o and 519 other objects, some of them in dlpp7M_tl_fc.a
14:56:57                       and 104 other libraries have wchar_t size 32 bits
14:56:57 [ERROR] Error[Li060]: module "cb_main.o(libublox-odin-w2-driver.a)" contains C++ code
14:56:57           built for use with the (E)C++98 library. This code is not link
14:56:57           compatible with the new (C++14) library.
14:56:57 Error[Li009]: runtime model conflict: Module
14:56:57           cb_main.o(libublox-odin-w2-driver.a) specifies that '__CPP_Language'
14:56:57           must be 'C++', but module authcrypt.o has the value 'C++14'. Modules
14:56:57           compiled with C++14 are not link-compatible with modules that are
14:56:57           compiled with older C++.
14:56:57 Error[Li060]: module "cb_ticker_wrapper.o(libublox-odin-w2-driver.a)" contains
14:56:57           C++ code built for use with the (E)C++98 library. This code is not
14:56:57           link compatible with the new (C++14) library.
14:56:57 Warning[Lt009]: Inconsistent wchar_t size
14:56:57             cb_main.o(libublox-odin-w2-driver.a) and 106 other objects have
14:56:57                       wchar_t size 16 bits
14:56:57             authcrypt.o and 519 other objects, some of them in dlpp7M_tl_fc.a
14:56:57                       and 104 other libraries have wchar_t size 32 bits
14:56:57 
14:56:57 [mbed] ERROR: "python" returned error code 1.
14:56:57 [mbed] ERROR: Command "python -u C:\builds\ws\mbedtls-mbed-os-examples-pr\authcrypt\mbed-os\tools\make.py -t IAR -m UBLOX_EVK_ODIN_W2 --source . --build .\BUILD\UBLOX_EVK_ODIN_W2\IAR" in "C:\builds\ws\mbedtls-mbed-os-examples-pr\authcrypt"
14:56:57 ---
```

andresag01, 2018-12-05:
I think this is an issue with Mbed OS, so I created the ticket https://github.com/ARMmbed/mbed-os/issues/8984 in that repository.

cmonr, 2018-12-06:
Fyi, Mbed OS is still not compatible with IAR 8.*
- https://os.mbed.com/docs/v5.10/tools/configuring-mbed-cli.html
- https://os.mbed.com/docs/v5.10/tools/exporting.html

andresag01, 2018-12-09:
@cmonr: Thanks for the information.

@sbutcher-arm @RonEld: Can we close this ticket? Perhaps we should mark it 'wont fix'

RonEld, 2018-12-10:
I think we should close this, at least until Mbed OS will support IAR 8

closed 2018-12-10 by RonEld