ARMmbed/mbed-os-example-tls #142: Application do not compile

mapellil, 2018-01-09:
the easy-connect generates compile errors when compiled for  target NUCLEO_L476RG + WIFI_IDW01M1 network interface. The CLI build system  and the online compiler both generates compiler errors.
The online compiler throw this error:

Error: Cannot open source input file "SpwfSAInterface.h": No such file or directory in "easy-connect/easy-connect.h", Line: 32, Col: 30

the CLI:

$ mbed compile
Building project mbed-os-example-tls-tls-client (NUCLEO_L476RG, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_COMMON_PAL
Compile [  0.3%]: BufferedPrint.c
Compile [  0.6%]: MyBuffer.cpp
Compile [  0.9%]: BufferedSerial.cpp
Compile [  1.1%]: main.cpp
[Fatal Error] easy-connect.h@32,29: SpwfSAInterface.h: No such file or directory
[ERROR] In file included from .\main.cpp:36:0:
./easy-connect/easy-connect.h:32:29: fatal error: SpwfSAInterface.h: No such file or directory
 #include "SpwfSAInterface.h"
                             ^
compilation terminated.

RonEld, 2018-07-26:
@mapellil I apologize for delayed reply.
Does this issue still reproduce?
Have you tried running `mbed deploy` ?

sbutcher-arm, 2018-10-31:
This is an old question, not a confirmed bug that's quite old. I assume this issue is now handled, so closing.

closed 2018-10-31 by sbutcher-arm