ARMmbed/mbed-os-example-tls #152: OOB-5.8 Does not compile within online compiler

donatieng, 2018-03-06:
Tried with K66F & NUCLEO_F429ZI targets.

Looks like there's a define issue with easy-connect.

> Error: #error directive: "No connectivity method chosen. Please add 'config.network-interfaces.value' to your mbed_app.json (see README.md for more information)." in "tls-client/easy-connect/easy-connect.h", Line: 49, Col: 3

These work fine with GCC_ARM + mbed cli, might be an online compiler issue.

mazimkhan, 2018-03-08:
I can confirm this does not happen on host. May be only online compiler issue. Assigning to @ashok-rao to route to concerned team.

adbridge, 2018-03-21:
@theotherjimmy  Could you take a look please ?

theotherjimmy, 2018-03-21:
Well, for starters, that config error is done incorrectly, and the text is wrong.

theotherjimmy, 2018-03-21:
@donatieng Did you try adding the value requested to your `mbed_app.json`? (I know it's obvious, but it's worth checking)