ARMmbed/mbed-os-example-tls #153: MAKE_ARM5 export compilation fails for NUCLEO_F429ZI

donatieng, 2018-03-07:
Using benchmark example:

```shell
mbed export -i MAKE_ARMC5 -m NUCLEO_F429ZI
...
make -j4
...

"link: benchmark.elf"
"C:\Users\dongar01\AppData\Local\Temp\p1b14-3", line 117 (column 3): Warning: L6312W: Empty Execution region description for region RW_IRAM1
Error: L6218E: Undefined symbol Load$$LR$$LR_IROM1$$Base (referred from mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/system_init_pre.o).
Finished: 0 information, 1 warning and 1 error messages.
make[1]: *** [benchmark.elf] Error 1
make: *** [all] Error 2
```

theotherjimmy, 2018-03-08:
Does this affect other targets? I'll try this later today.

theotherjimmy, 2018-03-08:
@donatieng Did you clean since you last built with a Makefile?

donatieng, 2018-03-09:
Had done a make clean but this time I started again with a clean checkout and it worked, let me try and make it fail again!

donatieng, 2018-03-09:
Closing for now - must have missed something or my Windows VM did something unexpected.

closed 2018-03-09 by donatieng