ARMmbed/mbed-os-example-tls #157: [OoB_5.8]: Need more docs.

ashok-rao, 2018-03-20:
While OoB campaign for 5.8 release, the feature opaque key was validated.
Docs start here: https://github.com/ARMmbed/mbed-os-example-tls/blob/feature-opaque-keys/README.md .
Observations:
1. No mention of opaque key feature.
2. No introduction to what the feature does, how to use it, what are the advantages etc.
3. On the individual examples (pk_sign, pk_verify) there is no info about how opaque keys are actually being used in these examples.
4. Currently HW support limited to K64 via source code. Porting to other platforms or use of the AT crypto engine with other platforms must be clearly explained (I2C pins to check, define if required etc) in a separate section.

The examples build / run fine on the CLI with all 3 supported toolchains.

cc @mazimkhan @sbutcher-arm

sbutcher-arm, 2018-07-25:
@Patater - The feature of opaque key handling was never released beyond a demo, and to my knowledge has been superseded by other planned work.

Therefore - as we're never going to release it, can we close this bug?

Patater, 2018-08-10:
The opaque keys work will be delivered differently as part of [Mbed Crypto](https://github.com/ARMmbed/mbed-crypto). Closing.

closed 2018-08-10 by Patater