ARMmbed/mbed-os-example-tls #159: [OoB_5.8]: export to make files does not build.

ashok-rao, 2018-03-20:
Steps:
1. mbed export -i make_gcc_arm -m K64F
2. Navigate to root of example.
3. make clean
4. make

Expected result:
Makefile builds binary of example.

Actual result:
Compilation fail.
```
C:\Ashok\mbed_CLI\OoB_5_8\TLS\mbed-os-example-tls>make
C:/Ashok/mbed_CLI/OoB_5_8/TLS/mbed-os-example-tls/makefile:805: warning: overriding commands for target `.s.o'
C:/Ashok/mbed_CLI/OoB_5_8/TLS/mbed-os-example-tls/makefile:797: warning: ignoring old commands for target `.s.o'
"Compile: main.c"
../pk_verify/main.c:23:28: fatal error: mbedtls/config.h: No such file or directory
 #include "mbedtls/config.h"
                            ^
compilation terminated.
make[1]: *** [../pk_verify/main.o] Error 1
make: *** [all] Error 2

C:\Ashok\mbed_CLI\OoB_5_8\TLS\mbed-os-example-tls>
```

Makefile attached (remove txt extension - only used for uploading to GitHub).

[Makefile.txt](https://github.com/ARMmbed/mbed-os-example-tls/files/1830427/Makefile.txt)

ashok-rao, 2018-03-28:
@theotherjimmy ..FYI. Could you please have a look? Thanks.

theotherjimmy, 2018-03-28:
@ashok-rao That makefile contains an extra level of `../` how did you export it? You should be running the export from the root of an example, so step 2 should not happen.

ashok-rao, 2018-03-28:
TLS repo contains a set of examples.. I was on the pk_sign example here: https://github.com/ARMmbed/mbed-os-example-tls/tree/feature-opaque-keys .. I ran export & didn't get the makefile in "that" example.. so I moved 1 level up to the root of TLS and found the Makefile there.. that's what I then "made" with make ..

theotherjimmy, 2018-03-28:
@ashok-rao Why is that in quotes? 

----

It looks like Mbed CLI does not know where your example starts because there are no `.mbed` files in this repo. I'm really not sure what's supposed to happen here. try placing a `.mbed` file in the root of the project and then exporting again.

ashok-rao, 2018-03-28:
:-) .. "that" above referred to pk_sign example among the many others.. hence.. 
Ok.. will try placing a .mbed file in the example (not the root) and try again..

sbutcher-arm, 2018-07-25:
Looks like an issue for the documentation?