ARMmbed/mbed-os-example-tls #163: Feature request : UDP support

JanneKiiskila, 2018-04-05:
As part of https://github.com/ARMmbed/mbed-os/issues/6545 investigations we come to the conclusion that we have a test gap - the TLS connections are not tested with UDP (with the exception of Mbed Cloud Client) - so as such it would be really good to have both TCP and UDP as options for the TLS example.

Could it be added? It should be preferably an option you can configure from the `mbed_app.json` file.

sbutcher-arm, 2018-07-25:
The purpose of these samples are to demonstrate the technology and use of the API. Testing is covered elsewhere.

@JanneKiiskila - Do you just want a DTLS system test for an Mbed OS board? If so, that's something else which we're already working on, and is nothing to do with this sample code.

cc: @mazimkhan

JanneKiiskila, 2020-10-26:
@adbridge  - seems to be me the BOT is teasing me for no reason, the ticket was updated.