ARMmbed/mbed-os-example-tls #171: MBEDTLS_MEMORY_BUFFER_ALLOC_C feature in benchmark is problematic

RonEld, 2018-05-23:
The `MBEDTLS_MEMORY_BUFFER_ALLOC_C` feature in the `benchmark` application is problematic, as it uses 64 KB of stack size. By default Mbed-OS uses 4 KB of stack size. This feature should either be removed, or length should be taken from the link script

sbutcher-arm, 2018-07-25:
I don't think anyone using Mbed OS should need or want to use `MBEDTLS_MEMORY_BUFFER_ALLOC_C`. Why did this come up? My automatic answer would be don't do it. (And maybe we should document that).

RonEld, 2018-07-26:
This came up in [this comment](https://github.com/ARMmbed/mbed-os-example-tls/pull/116#discussion_r188403649)

I agree that anyone using Mbed OS should need to use `MBEDTLS_MEMORY_BUFFER_ALLOC_C` , this is why the descriptionsais:
>This feature should either be removed, or length should be taken from the link script

sbutcher-arm, 2018-07-26:
Ah! I was being dumb. I didn't realise you meant it was in `[benchmark.cpp](https://github.com/ARMmbed/mbed-os-example-tls/blob/master/benchmark/main.cpp)`. I thought you were talking about the library. That should indeed be removed.

RonEld, 2018-07-26:
no problem, I can see how me description was confusing.

RonEld, 2018-07-26:
After looking some more, I am not sure we want to remove the call for `MBEDTLS_MEMORY_BUFFER_ALLOC_C` in `benchmark.c` The feature is also used for memory statistics, measured in `MEMORY_MEASURE_PRINT`, which we may want to utilize for debugging

RonEld, 2018-07-30:
@andresag01 Yes, Mbed-OS has its own facilities, but you can't differentiate between Mbed TLS memoey usage and other features.
I am aware that `MBEDTLS_MEMORY_BUFFER_ALLOC_C` was probably not used in Mbed-OS, but I was raising a thought that perhaps we would like to keep it, for debugging purposes.
Obviously, the buffer size should change, perhaps according to scatter list

andresag01, 2018-07-31:
@RonEld

>Yes, Mbed-OS has its own facilities, but you can't differentiate between Mbed TLS memory usage and other features.

I would definitely consider this if we were talking about a relatively large application with multiple threads and other modules (other than Mbed TLS) being used. However, the benchmark example is just a simple application running Mbed TLS tasks sequentially (and nothing else) and there should be almost no interference wrt performance or memory from other parts of the system. I personally think that we should aim to simplify this examples and reduce the maintenance cost instead of adding to it.

> I am aware that MBEDTLS_MEMORY_BUFFER_ALLOC_C was probably not used in Mbed-OS, but I was raising a thought that perhaps we would like to keep it, for debugging purposes.
Obviously, the buffer size should change, perhaps according to scatter list

If we really want to keep this and if you mean "scatter file" as in linker script, then I would prefer to not rely on this as I am not sure that different platforms export the same symbols. I think that we probably just need to use some Mbed OS facilities to check how much memory there is (if possible) and then dynamically allocate a huge buffer using regular malloc, then configure the allocator.

andresag01, 2018-10-18:
@sbutcher-arm, @RonEld: I think this was fixed by https://github.com/ARMmbed/mbed-os-example-tls/pull/194. Should we close this issue

andresag01, 2018-10-24:
This was fixed via #194, so it can now be closed.

closed 2018-10-24 by andresag01