ARMmbed/mbed-os-example-tls #183: Feature request - Provide a TLS Server example

sbutcher-arm, 2018-07-25:
Some users would like to use Mbed OS to provide TLS server functionality, and would like an example to help them. This [has been requested on the Mbed OS repo](https://github.com/ARMmbed/mbed-os/issues/4670).

sbutcher-arm, 2020-11-20:
I'm afraid I don't work on this project any more, and I don't know if this feature is needed any longer, so I am not the best person to complete this.

simonbutcher, 2020-12-08:
For the sake of stopping the annoying @ciarmcom bot, that just wants a 25 word answer to shut up....

**Could you add some more detail to the description?** 

Some users asked for an example that shows how to provide a TLS server. How that is provided is down to the person providing it.

Worth noting though that from a security and power management perspective, client is better than server in IOT. Server needs to be always on, always connected, and always available, so the far end can poll it. That means an open port always open to an attacker, and the device to be perpetually powered up and waiting, and potentially a radio too, if it's wireless.

That's why we prefer client.

**What target(s) are you using?**

n/a

**What toolchain(s) are you using?**

n/a

**What version of Mbed OS are you using (tag or sha)?**

n/a

**It would help if you could also specify the versions of any tools you are using?
How can we reproduce your issue?**

n/a

0xc0170, 2020-12-14:
There is a fix, it should fix. Sorry about the noise @simonbutcher