ARMmbed/mbed-os-example-tls #184: Benchmark example fails on K64F with ARM C Compiler

sbutcher-arm, 2018-07-25:
The Benchmark example is failing at the ECDSA test (again) on K64F and NUCLEO_F429ZI with the ARM C Compiler.

This is a regression introduced by PR #116. 

We need to revert this change.

cc: @k-stachowiak, @RonEld

RonEld, 2018-07-26:
I have just tested again the benchmark application, compiled with ARM toolchain, and debug profile, on a k64f, and didn't encounter any issue:
~~~
  SHA-256                  :        732 KB/s
  SHA-512                  :        551 KB/s
  AES-CBC-128              :        657 KB/s
  AES-CBC-192              :        562 KB/s
  AES-CBC-256              :        491 KB/s
  AES-GCM-128              :        352 KB/s
  AES-GCM-192              :        323 KB/s
  AES-GCM-256              :        298 KB/s
  AES-CCM-128              :        293 KB/s
  AES-CCM-192              :        254 KB/s
  AES-CCM-256              :        225 KB/s
  AES-CMAC-128             :        589 KB/s
  AES-CMAC-192             :        512 KB/s
  AES-CMAC-256             :        450 KB/s
  AES-CMAC-PRF-128         :        588 KB/s
  CTR_DRBG (NOPR)          :        482 KB/s
  CTR_DRBG (PR)            :        357 KB/s
  HMAC_DRBG SHA-256 (NOPR) :         87 KB/s
  HMAC_DRBG SHA-256 (PR)   :         77 KB/s
  RSA-2048                 :      58 ms/ public
  RSA-2048                 :    2346 ms/private
  RSA-4096                 :     201 ms/ public
  RSA-4096                 :   13190 ms/private
  ECDSA-secp384r1          :     777 ms/sign
  ECDSA-secp256r1          :     492 ms/sign
  ECDSA-secp384r1          :    1494 ms/verify
  ECDSA-secp256r1          :     950 ms/verify
  ECDHE-secp384r1          :    1435 ms/handshake
  ECDHE-secp256r1          :     906 ms/handshake
  ECDHE-Curve25519         :     791 ms/handshake
  ECDH-secp384r1           :     702 ms/handshake
  ECDH-secp256r1           :     453 ms/handshake
  ECDH-Curve25519          :     401 ms/handshake

DONE
~~~

sbutcher-arm, 2018-07-26:
Hi @RonEld,

This came up when I did a release job on the development head. You can see the failure for yourself in build #611 on Jenkins. What version of Mbed TLS were you using? Can you retest using commit id f11a7cda73e4?

RonEld, 2018-07-26:
I was testing on latest master branch of `mbed-os-example-tls` and `mbed-OS` which uses `mbedtls-2.10.0`
I'll retest

RonEld, 2018-07-26:
Hi @sbutcher-arm Could you please refer me to what repository is commit id f11a7cda73e4?
In addition, could you please refer me to the specific Jenkins release job? I was looking at [this failed job](https://jenkins-internal.mbed.com/job/mbedtls-release/611/) and I don't think it's related.

RonEld, 2018-07-26:
I got my answers. The repository is `mbedtls-restricted` and that was the Jenkins job.
I have change the MBED_TLS_VERSION in the importer script to f11a7cda73e4 and imported the code.
I compiled with GCC_ARM toolchain, with debug profile, for K64f.
The issue was not reproduced on my test:
~~~
  SHA-256                  :        515 KB/s
  SHA-512                  :        333 KB/s
  AES-CBC-128              :        498 KB/s
  AES-CBC-192              :        435 KB/s
  AES-CBC-256              :        387 KB/s
  AES-GCM-128              :        195 KB/s
  AES-GCM-192              :        185 KB/s
  AES-GCM-256              :        175 KB/s
  AES-CCM-128              :        229 KB/s
  AES-CCM-192              :        202 KB/s
  AES-CCM-256              :        180 KB/s
  AES-CMAC-128             :        445 KB/s
  AES-CMAC-192             :        393 KB/s
  AES-CMAC-256             :        352 KB/s
  AES-CMAC-PRF-128         :        444 KB/s
  CTR_DRBG (NOPR)          :        401 KB/s
  CTR_DRBG (PR)            :        292 KB/s
  HMAC_DRBG SHA-256 (NOPR) :         63 KB/s
  HMAC_DRBG SHA-256 (PR)   :         56 KB/s
  RSA-2048                 :     173 ms/ public
  RSA-2048                 :    6865 ms/private
  RSA-4096                 :     638 ms/ public
  RSA-4096                 :   41656 ms/private
  ECDSA-secp384r1          :    1521 ms/sign
  ECDSA-secp256r1          :     831 ms/sign
  ECDSA-secp384r1          :    2926 ms/verify
  ECDSA-secp256r1          :    1608 ms/verify
  ECDHE-secp384r1          :    2828 ms/handshake
  ECDHE-secp256r1          :    1557 ms/handshake
  ECDHE-Curve25519         :    1358 ms/handshake
  ECDH-secp384r1           :    1389 ms/handshake
  ECDH-secp256r1           :     768 ms/handshake
  ECDH-Curve25519          :     696 ms/handshake

DONE
~~~

RonEld, 2018-07-31:
I just noticed this issue was raised on ARM C toolchain.
When I tried building for ARM C toolchain, I got a build failure. Raised as https://github.com/ARMmbed/mbedtls/issues/1907

I fixed the build issue. But I still didn't reproduce the current issue:
~~~
  SHA-256                  :        515 KB/s
  SHA-512                  :        333 KB/s
  AES-CBC-128              :        498 KB/s
  AES-CBC-192              :        435 KB/s
  AES-CBC-256              :        387 KB/s
  AES-GCM-128              :        195 KB/s
  AES-GCM-192              :        185 KB/s
  AES-GCM-256              :        175 KB/s
  AES-CCM-128              :        229 KB/s
  AES-CCM-192              :        202 KB/s
  AES-CCM-256              :        180 KB/s
  AES-CMAC-128             :        445 KB/s
  AES-CMAC-192             :        393 KB/s
  AES-CMAC-256             :        352 KB/s
  AES-CMAC-PRF-128         :        444 KB/s
  CTR_DRBG (NOPR)          :        401 KB/s
  CTR_DRBG (PR)            :        292 KB/s
  HMAC_DRBG SHA-256 (NOPR) :         63 KB/s
  HMAC_DRBG SHA-256 (PR)   :         56 KB/s
  RSA-2048                 :     173 ms/ public
  RSA-2048                 :    6865 ms/private
  RSA-4096                 :     638 ms/ public
  RSA-4096                 :   41656 ms/private
  ECDSA-secp384r1          :    1521 ms/sign
  ECDSA-secp256r1          :     831 ms/sign
  ECDSA-secp384r1          :    2926 ms/verify
  ECDSA-secp256r1          :    1608 ms/verify
  ECDHE-secp384r1          :    2828 ms/handshake
  ECDHE-secp256r1          :    1557 ms/handshake
  ECDHE-Curve25519         :    1358 ms/handshake
  ECDH-secp384r1           :    1389 ms/handshake
  ECDH-secp256r1           :     768 ms/handshake
  ECDH-Curve25519          :     696 ms/handshake

DONE
~~~

Nonetheless, I will remove the ECDSA build, as the CI probably fails with it, and it shouldfail.

RonEld, 2018-08-01:
Tested also with ARMC toolchain, and no issue found:
~~~
  SHA-256                  :        731 KB/s
  SHA-512                  :        551 KB/s
  AES-CBC-128              :        651 KB/s
  AES-CBC-192              :        557 KB/s
  AES-CBC-256              :        486 KB/s
  AES-GCM-128              :        350 KB/s
  AES-GCM-192              :        320 KB/s
  AES-GCM-256              :        295 KB/s
  AES-CCM-128              :        290 KB/s
  AES-CCM-192              :        252 KB/s
  AES-CCM-256              :        222 KB/s
  AES-CMAC-128             :        585 KB/s
  AES-CMAC-192             :        507 KB/s
  AES-CMAC-256             :        446 KB/s
  AES-CMAC-PRF-128         :        585 KB/s
  CTR_DRBG (NOPR)          :        477 KB/s
  CTR_DRBG (PR)            :        354 KB/s
  HMAC_DRBG SHA-256 (NOPR) :         89 KB/s
  HMAC_DRBG SHA-256 (PR)   :         78 KB/s
  RSA-2048                 :      58 ms/ public
  RSA-2048                 :    2343 ms/private
  RSA-4096                 :     201 ms/ public
  RSA-4096                 :   13203 ms/private
  ECDSA-secp384r1          :     778 ms/sign
  ECDSA-secp256r1          :     487 ms/sign
  ECDSA-secp384r1          :    1503 ms/verify
  ECDSA-secp256r1          :     945 ms/verify
  ECDHE-secp384r1          :    1439 ms/handshake
  ECDHE-secp256r1          :     910 ms/handshake
  ECDHE-Curve25519         :     788 ms/handshake
  ECDH-secp384r1           :     706 ms/handshake
  ECDH-secp256r1           :     451 ms/handshake
  ECDH-Curve25519          :     399 ms/handshake

DONE
~~~

sbutcher-arm, 2018-10-18:
Fixed by PR #194 which has now been merged. Issue can be closed.

closed 2018-10-18 by sbutcher-arm