ARMmbed/mbed-os-example-tls #210: Benchmark example times out on KW24D, on ARM compiler

k-stachowiak, 2018-10-31:
The example times out while performing benchmark for `DHE-2048`. Surrounding log:

```
[1540134513.96][CONN][RXD]   RSA-4096                 :   20077 ms/private
[1540134527.98][HTST][INF] test suite run finished after 600.77 sec...
[1540134527.98][HTST][ERR] Expected output [\s+DHE-2048\s*:\s*\d+ ms/handshake] not received in log.
```

k-stachowiak, 2018-10-31:
Note that it is similar to #208

RonEld, 2019-01-03:
@k-stachowiak Does the example timeout when setting `MBEDTLS_MPI_MAX_SIZE` to 512 ?

k-stachowiak, 2019-01-17:
@sbutcher-arm @Patater Based on #231 I conclude, that the problem doesn't occur anymore. I suggest closing this issue.

k-stachowiak, 2020-11-09:
I still think we can close it. Let me free some @ciarmcom 's cycles and close it.

closed 2020-11-09 by k-stachowiak